pvvx/ATC_MiThermometer #455: [feature request]

txitxo0, 2023-12-30:
Hi there, I love this project! And one of the last majors being able to migrate to ZigBee blow my mind!

Having the possibility to add a relay I have been thinking in a thermostat, triggering this relay based on the temperature the device measures an a target temperature. Could be possible (or in your road map) to add a possibility to display this target temperature? 

If it is possible to dinamically set this target temperature out from home assistant would be the ultimate cheapest thermostat in the market!

pvvx, 2023-12-31:
https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#temperature-or-humidity-trigger-gpio-pa5-lywsd03mmc-label-on-the-reset-pin

txitxo0, 2024-01-01:
Thank you to answer so fast always and happy new year!

Yes, but as long as I get from the docs, the target temperature is not being shown in the display, is it?

pvvx, 2024-01-01:
Happy new year!
The thermometer has long been fulfilling all your requests with the functions it contains.
But I'm not the creator of Home Assistant.

txitxo0, 2024-01-01:
> Happy new year!
> The thermometer has long been fulfilling all your requests with the functions it contains.
> But I'm not the creator of Home Assistant.

Thank you! I will try to re-read carefully to get knowledge about this two features
- display target temperature
- dinamically set the target temperature up

Regards!