pvvx/ATC_MiThermometer #458: [Feature request] Show battery only if <20%

AleXSR700, 2024-01-03:
Hi,
I could have sworn this used to be normal behavior, but would it be possible to add the option to show battery level on the screen if it drops below a certain value (e.g. 20 % or 25 %)?

The information is not needed as long as battery is full, but once it heads towards a certain level, BLE will stop working. Being able to see low battery warning would be great.

Thank you
Alex

pvvx, 2024-01-03:
[measured_data.battery_level <= 5](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd.c#L89C8-L89C40)

AleXSR700, 2024-01-03:
> [measured_data.battery_level <= 5](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd.c#L89C8-L89C40)

Is this <= 5%?
If so, I think the threshold is too small as bluetooth will stop working long before that.

pvvx, 2024-01-04:
It works for me at 0% for about another month.

AleXSR700, 2024-01-08:
Including bluetooth?

I will "reboot" the device by removing the battery and see if the ble broadcasts again.

pvvx, 2024-01-08:
[image]

AleXSR700, 2024-01-08:
?
Your charts only go to ~20% battery level, not 0%. What were you trying to show me? :)

pvvx, 2024-01-08:
At lower temperatures the battery percentage will be lower. And so on until zero.

AleXSR700, 2024-01-08:
But the issue is not the temperature. I was/am talking room temperature.

At room temperature the ble will no work at 0% battery. It will probably stop working somewhere between 0-20 % battery level. So it would be a nice feature if the battery percentage would start showing once the battery level drops below x %. That way the user can see that ble stopped working due to low battery.

pvvx, 2024-01-08:
I have many thermometers. And they all work at 0% battery for quite a long time.

pvvx, 2024-01-08:
> At room temperature the ble will no work at 0% battery. 

Usually at 0% battery and +25C it still works for about a week.

Do you want the program to analyze temperature, battery quality, the presence of a capacitor in the power supply and other parameters?

AleXSR700, 2024-01-08:
I am not sure that is the case when the battery is discharged at room temperature. But I will monitor and report back.

pvvx, 2024-01-08:
0% battery = 2.2V. The chip operates up to 1.8V. Software limited to 2.0 V.

pvvx, 2024-01-08:
> I am not sure that is the case when the battery is discharged at room temperature. But I will monitor and report back.

If a capacitor is not soldered into the thermometer, then some batteries will work up to 20% and immediately turn off. Loss of battery charge in the absence of a capacitor recommended by the manufacturers in the power supply is up to 40%.

AleXSR700, 2024-01-13:
Interesting.
Any way of checking without breaking?

Mine, according to my data, dropped quite suddenly from 8% to 2%, stopped working and is showing:

[image]

Is that number supposed to represent something?

pvvx, 2024-01-13:
This is the voltage in mV. The recorded value is below 2.0 V, at which the thermometer goes into sleep mode for a few minutes.

edmundo096, 2024-01-25:
The info on the last comments are actually interesting features of the CFW.

Not sure if those "facts" are commented on the main README but they are quite nice:

>Usually at 0% battery and +25C it still works for about a week.

> 0% battery = 2.2V. The chip operates up to 1.8V. Software limited to 2.0 V.

>If a capacitor is not soldered into the thermometer, then some batteries will work up to 20% and immediately turn off. Loss of battery charge in the absence of a capacitor recommended by the manufacturers in the power supply is up to 40%.
> When there is a pulse load during transmission, the battery voltage drops and the thermometer turns off.

> The recorded value is below 2.0 V, at which the thermometer goes into sleep mode for a few minutes.
_...and LCD shows the mV._