pvvx/ATC_MiThermometer #459: Cannot connect to device via TelinkMiFlasher website for update and configuration

Robert-Versteegen, 2024-01-03:
Tried to update a new device (hardware v1.5  LYWSD03MMC with the flasher url but the device would not connect. 
Flashing the device with the atc1441 flasher url would connect instant and i'm able to flash the ATC_v46.bin file, and back to stock mi v130 firmware.

From the pvvxflasher site no connection. not from an edge browser on win11, nor from a android chrome browser. Device is listed in the devices, but unable to connect. Unable to connect to other currently working devices as wel.  

Done a number of these devices in the past. for update and configuration. 
Due to the fact i'm not able to connect, i cannot set the device type to the BTHome. 

Is there an issue with the flasher? or am i the only one

pvvx, 2024-01-03:
Everything is working. Just tested on different options.

Robert-Versteegen, 2024-01-03:
thanks for testing.. so it must be something here.

sanosis, 2024-01-08:
Arch Linux - current, bluez 5.71-3, chromium 120.0.6099.199-1 The process is stuck on 'Connecting to:'

Robert-Versteegen, 2024-01-08:
Hi,  sometimes i still get the "connecting to:"... also.. What i do is i use the OTA flash url in another tab  This connects the device succesfully.. then in the flasher URL i can connect to the device also and do my update and config.

KlausHans, 2024-04-08:
Got heavy connection problems her too. I am using Chromium on the latest Ubuntu.
[image]
I was able to flash the firmware and already connected to it afterwards. Then the connection seem lost so i tried to reconnect. No success. Now i am trying since 10 minutes to connect again. Other tab didn't helped. Bluetooth disabled and enabled on the pc didn't helped. Battery reinserting in the Xiaomi device didn't helped.

Robert-Versteegen, 2024-04-09:
Hi, some late comment about my config. tried it mostly from my android. now it runs on v14. 
it resolved itself, and i can now use the flasher url directly. phone update or site change, cant say.
from my point of view this issue can be closed

pvvx, 2024-04-10:
> Ok, thanks for the info. Is there a way to reset the firmware without connecting to it?

In most cases, the advertising interval of 10 seconds is served by smartphones with Android. Try connecting and reducing the interval.

[README](https://github.com/pvvx/ATC_MiThermometer):
_If you have a smartphone with BT5.0+ and the nRFConnect program, then you can reset all settings to default by connecting to a thermometer and sending the 0x56 command, as shown in the screenshot:_