pvvx/ATC_MiThermometer #461: Screen off option?

Digital-Ark, 2024-01-04:
Hardware Version: LYWSD03MMC B1.7
Software Version: 4.6

Checking the screen off box seems to no longer update the screen, but it doesn't go blank.

Perhaps I don't understand correctly, and not updating the screen saves power.  I was hoping the whole screen would be blank.

My use case is a freezer sensor, where the screen is too cold to operate correctly.

pvvx, 2024-01-04:
After enabling the option, you need to restart the thermometer using power. Remove and insert the battery.
There is no documentation for the LCD display controller - only this shutdown option is used.

Digital-Ark, 2024-01-05:
Excellent, thank you.

I have 3D printed a case that uses 2 "AA" batteries instead of the CR2032 for my use case, it's working well.

closed 2024-03-09 by pvvx