pvvx/ATC_MiThermometer #465: Battery Drain on Xiaomi Mijia (LYWSD03MMC) with Zigbee Version of firmware.

mrkaqz, 2024-01-13:
I have flashed 2 of my Xiaomi Mijia LYWSD03MMC with the Telink Flasher 7.4 with Zigbee radio.
Then paried it with Z2MQTT on my HA.

After using for a while notice that the batter was low, then I changed to new battery still observed the drainage of battery.

Anyone experience the same?

[image]

[image]

Routout, 2024-01-17:
Yes, Battery last only couple of weeks. Going to revert back to BT.

pvvx, 2024-01-17:
https://github.com/pvvx/ZigbeeTLc/issues/37

closed 2024-01-17 by pvvx