pvvx/ATC_MiThermometer #466: Диапазон смещения влажности

DedkovArtem, 2024-01-14:
Имею в наличии несколько датчиков LYWSD03MMC у которых необходимо установить коррекцию влажности больше чем 12,7%. Насколько критично корректировать влажность с точностью до 0,1%? Предлагаю сделать её целочисленной

pvvx, 2024-01-14:
> Имею в наличии несколько датчиков LYWSD03MMC у которых необходимо установить коррекцию влажности больше чем 12,7%.

Испорченных датчиков?

pvvx, 2024-01-14:
Датчики влажности деградируют. В основном от загрязнений. А если новые - возможно попал флюс при пайке.
Восстановить невозможно. И у датчика будет другая диаграмма линейности и т.д. Т.е. необходимо корректировать и коэф. наклона (множитель) и смещение. Но это всё равно не поможет.

DedkovArtem, 2024-01-15:
Спасибо за быстрый ответ. Думаю действительно будет проще заменить на новый.

closed 2024-01-15 by DedkovArtem

pvvx, 2024-01-15:
Два Xiaomi LYWSD03MMC B1.9 и B1.4 на одной полке.  
Один с прошивкой Zigbee, другой c BLE. 
Даты выпуска имеют разницу в 1 год.
Никаких коррекций.
[image]
[image]
Но врут, т.к. по расчету точки росы у них разница:
[image]