pvvx/ATC_MiThermometer #467: LYWSD02MMC support

Avalonnw, 2024-01-16:
Hi,
I've got a LYWSD02MMC version (with bigger display) however when I try to connect to it I get message: `Not found Telink OTA service!`
Is this device not supported? Is there a chance for its support in the future?

pvvx, 2024-01-17:
This repository is called: 

## Custom firmware for BLE thermometers **on the Telink chipset**.

https://pvvx.github.io/LYWSD02/

closed 2024-01-17 by pvvx

tbaumann, 2024-01-19:
> This repository is called:
> ## Custom firmware for BLE thermometers **on the Telink chipset**.
> 
> https://pvvx.github.io/LYWSD02/

What do you mean by that? Currently only supported by vendor firmware?

pvvx, 2024-01-20:
This device has a different chipset. I do not do it.