pvvx/ATC_MiThermometer #469: Cannot activate in MiHome app after flashing firmware

pigl3t-mahdi, 2024-01-18:
I have flashed the latest 4.6 version successfully. Everything seems to work fine, but when activating in MIHome app i get:  "standard verification failed  (-27)" error. Is this any check that Xiaomi is doing to disallow custom firmware? Anything an be done ?

pvvx, 2024-01-18:
[README](https://github.com/pvvx/ATC_MiThermometer#)
_Custom firmware is not supported in MiHome._

closed 2024-01-18 by pvvx

mcbyte-it, 2025-03-16:
But in the readme it says:
_Custom firmware is not supported in MiHome. But it is possible to work LYWSD03MMC with 'Xiaomi Gateway 3' by first registering in MiHome on the official firmware, flashing to custom firmware with restoring the binding key from MiHome, replacing the device name with "LYWSD03MMC" and setting the advertising type to "MIJIA (MiHome)" marked "encrypted"._

I've done all this, and it gives the error as @pigl3t-mahdi says