pvvx/ATC_MiThermometer #470: CGG1-M 2022 shows 0,0°C temperature and 0% humidity

TheZoker, 2024-01-19:
Hi there,

I have a CGG1-M 2022 with version 4.6 flashed. 
It worked normal for the last couple of weeks, but starting some days ago, it showed and send the same temperature over and over again. I then removed the battery and but a new one in and since then it only shows 0,0°C for the temperature and 0% for the humidity. 

Is there anything I can do to fix this? 

[image]

[image]

pvvx, 2024-01-20:
So far I have not encountered such a malfunction...

airy10, 2024-01-20:
The only time I've seen some similar behavior was after one of my outside devices felt into some water (while it was a LYWSD03MMC)
Removing the battery and letting it dry for a few hours before putting the battery back fixed it.

TheZoker, 2025-03-16:
For further investigation I flashed the device back to the original firmware. It now shows E2, which according to this reddit thread means, that there might be corrosion or any other water damage to the device:
https://www.reddit.com/r/Xiaomi/comments/ez2edm/e2_on_mi_temperature_and_humidity_monitor_what/

closed 2025-04-23 by pvvx