pvvx/ATC_MiThermometer #473: LYWSD03MMC - Same Values since 2 Months

mlinko6, 2024-01-30:
Dear all,

I'm using 4 LYWSD03MMC Temp and Humidity Meters. 3 are flashed and connected over Bluetooth and one is flashed with Zigbee. Then last one worked fine for a while but the it just "froze" at a certain value:

Temp: 22,65
Humidity: 48.68

The only correct value is the Battery which is changing... Do you have any clues on what I did wrong or what I should do to "revoke" the sensor?

Is it possible for me to flash the Sensor with the bluetooth firmware?

Thank you and kind regards,
Rok

pvvx, 2024-01-31:
Support for Zigbee firmware is in the [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc) repository.

naimadfpv, 2024-03-06:
Im having the same issue since yesterday. 

I deleted one of the thermometer which did work for a very long time because i placed it in another room and i wanted to pair it with another repeater.

After succesfully reconnect to the new repeater it did not read temperature, humidity and battery values. it stays with the values after reconnection.

I tried several reconnects with z2m and also reseted the sensor. 
The sensor is not working anymore and always showing as offline in z2m. 
But when I read values manually in z2m it is reporting values from battery, humidity and temperature.
I can set it from celsius to farenheit in z2m GUI but i cannot disable the icon smiley and cannot hide display. 
I also cannot reconfigure the device.

I have read that i have to put lywsd03mmc-z.js in z2m config path and set external konverter in z2m. Some devices now show as lywsd03mmc-z this one not. it is showing as lywsd03mmc in z2m.

z2m gui shows that its seen but it does not report values.

Firmware Date: 20240211
Firmware Version: 3001-0120

Logs show that the device is reporting values:
`Debug 2024-03-06 07:00:20Received Zigbee message from '0xa4c138140e3a285a', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2096}' from endpoint 1 with groupID 0
Info 2024-03-06 07:00:20MQTT publish: topic 'zigbee2mqtt/0xa4c138140e3a285a', payload '{"battery":97,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"enable_display":null,"humidity":38.82,"humidity_calibration":null,"last_seen":"2024-03-06T07:00:20+01:00","linkquality":94,"show_smiley":"HIDE","temperature":20.96,"temperature_calibration":null,"temperature_display_mode":"fahrenheit","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2024-03-06 07:00:40Received Zigbee message from '0xa4c138140e3a285a', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":3888}' from endpoint 1 with groupID 0
Info 2024-03-06 07:00:40MQTT publish: topic 'zigbee2mqtt/0xa4c138140e3a285a', payload '{"battery":97,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"enable_display":null,"humidity":38.88,"humidity_calibration":null,"last_seen":"2024-03-06T07:00:40+01:00","linkquality":91,"show_smiley":"HIDE","temperature":20.96,"temperature_calibration":null,"temperature_display_mode":"fahrenheit","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'`

but the values do not change and the device is reported as offline.

[image]
if I refresh the values in GUI it is reporting values almost immediatly.

closed 2024-03-09 by pvvx