pvvx/ATC_MiThermometer #476: Error When Paring LYWSD03MMC with Z2MQTT

mrkaqz, 2024-02-01:
This error right after starting to interview.

2024-02-01 17:48:00Failed to configure '0xa4c1388366104b9b', attempt 2 (Error: Bind 0xa4c1388366104b9b/1 msTemperatureMeasurement from '0x00212effff07928c/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:545:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/data/extension/externally-loaded.js:17:9) at Configure.configure (/app/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))

Any idea what it is and how to fix it?

KennethLavrsen, 2024-02-11:
I am having the exact same error.

informagico, 2024-03-08:
Evening gents, I am having the same issue on all of my 3 T&H sensors.
HomeAssistant with zigbee2mqtt is pairing the device correctly but after the interview the device reports always this error

```log
Error 2024-03-08 17:35:25Failed to configure '0xa4c1380b92710c97', attempt 1 (Error: Bind 0xa4c1380b92710c97/1 msTemperatureMeasurement from '0x00212effff0982b9/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/app/data/extension/externally-loaded.js:125:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Info 2024-03-08 17:35:26MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254,"value":false},"status":"ok","transaction":"3m6r7-2"}'

Info 2024-03-08 17:35:36Configuring '0xa4c1380b92710c97'

Info 2024-03-08 17:35:36MQTT publish: topic 'zigbee2mqtt/0xa4c1380b92710c97', payload '{"battery":null,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"humidity":null,"humidity_calibration":null,"linkquality":255,"show_smiley":null,"temperature":18.01,"temperature_calibration":null,"temperature_display_mode":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'

Info 2024-03-08 17:35:36MQTT publish: topic 'zigbee2mqtt/0xa4c1380b92710c97', payload '{"battery":null,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"humidity":71.18,"humidity_calibration":null,"linkquality":255,"show_smiley":null,"temperature":18.01,"temperature_calibration":null,"temperature_display_mode":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'

Error 2024-03-08 17:35:37Failed to configure '0xa4c1380b92710c97', attempt 2 (Error: Bind 0xa4c1380b92710c97/1 msTemperatureMeasurement from '0x00212effff0982b9/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/app/data/extension/externally-loaded.js:125:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Info 2024-03-08 17:35:46Configuring '0xa4c1380b92710c97'

Info 2024-03-08 17:35:46MQTT publish: topic 'zigbee2mqtt/0xa4c1380b92710c97', payload '{"battery":null,"comfort_humidity_max":null,"comfort_humidity_min":null,"comfort_temperature_max":null,"comfort_temperature_min":null,"humidity":65.72,"humidity_calibration":null,"linkquality":255,"show_smiley":null,"temperature":18.01,"temperature_calibration":null,"temperature_display_mode":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'

Error 2024-03-08 17:35:47Failed to configure '0xa4c1380b92710c97', attempt 3 (Error: Bind 0xa4c1380b92710c97/1 msTemperatureMeasurement from '0x00212effff0982b9/1' failed (Error: status: 140) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at configure (/app/data/extension/externally-loaded.js:125:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

I am using a ConBeeII coordinator key

pvvx, 2024-03-09:
ConBeeII: [Failed to configure ... failed (Error: status: 140) at DeconzAdapter.bind ](https://www.google.com/search?q=Failed+to+configure++failed+%28Error+status+140%29+at+DeconzAdapter.bind)

KennethLavrsen, 2024-03-09:
Not sure what that Google search was meant to mean. But I do not use a Conbee stick. I use a Sonoff P stick. And when I switched to using the https://github.com/devbis/z03mmc firmware instead, the error reported here is gone. I repeated this on 4 devices. So devbis has something right that is missed in this version

pvvx, 2024-03-09:
You have a converter installed for the thermometer version from devbis.
You did not install the correct converter or did not update Z2M.