pvvx/ATC_MiThermometer #477: Good news for LYWSD03MMC  owners who need longer operation times

ilgrank, 2024-02-11:
Thanks for your hard work @pvvx !
For once, I'm not asking anything, but wanted to share some test I have made.
I am using LYWSD03MMC sensors in very cold weather (as you do too) and that kills battery life.
Add to this that long ago I asked if I could up the TX power and you replied that the CR2032 simply had not enough power to do that.
So I had the idea of using a larger and more capable battery: to make a long story short, I modified a CR2032 cell to house an high efficiency 3V voltage regulator, (with just 8uA quiescent current) and with a 18650 cell, I now had years of battery life even in -22°C!
[image]
But it gets even better! a LYWSD03MMC  costs just $4, so I decided to sacrifice one for science to check if could survive being fed 4.2V directly and if things could be made even simplier and.. to make a long story short, I hooked up the LYWSD03MMC  *directly* to a 18650 battery, and it worked!
[image]
[image]
It's being on since a couple of days at very low temp, and it's doing great so far

pvvx, 2024-02-11:
3941 mV is a lot. The microcircuit can degrade from such voltage and consume more than a new one.
I checked it on several chips. Once supply is exceeded, they begin to consume more and only replacement is possible.
The level of overvoltage at which degradation occurs depends on the chip. But it is higher than 3.8V...

ilgrank, 2024-02-11:
Thanks for your reply @pvvx !
I've put 3 versions to the test (I consider them expendable) one with 18650 and the regulator, one with the unregulated 18650 and now one with 18650 and 3 shottky diodes which gives about 3.3V to he LYWSD03MMC .
I've never wanted to recharge the battery in low temp btw, just avoid having to go and replace the CR2032 every 6 month as I did for the last years.
Given you have already done such tests.. will the power consumption rise much, like 2x at such (3.9V) high voltages? My lab equipment won't measure in the uA range, so the only measure I can have is voltage over time.

Also, thanks for the link, very informative!

pvvx, 2024-02-11:
> Given you have already done such tests.. will the power consumption rise much, like 2x at such (3.9V) high voltages?

After Over-Voltage, the consumption of the chip increases forever, at any other voltage. From several times to the point that the chip will heat up at 3V. But it continues to work... But it can also break through the crystal of the chip forever.

ilgrank, 2024-02-11:
Thanks for the quick reply!

> After Over-Voltage, 

According to your tests, what is the highest safe voltage? 3.6V is safe? (I'm thinking of 3xLSD NiMh)

Also, Your estimate of 1 year of battery out of 2xAAA points to a great reduction in useful capacity tho:  2xAAA should have roughtly 5x the capacity of a CR2032, so even if BLE is consuming more, it should last at least 2-3 years, no?

> No need for Schottky

That's what I had lying around :)

pvvx, 2024-02-11:
Nowadays they produce “lightweight” AAA batteries. Their capacity may be less good CR2032 :)

pvvx, 2024-02-12:
> According to your tests, what is the highest safe voltage?

3.8...4+ V - Depending on the chip instance.

ilgrank, 2024-02-12:
That graph is really worrysome! Worse than low-quality CR2032!
Being that my source is a Li battery, the potential for high current is enormous.. but luckily, I did not fully charge it, as it was at around 3975mV when I connected it maybe I've been lucky then .. we'll see :)

Another idea I tried in the past is using CR2050 batteries: they do fit in with very minor modification and have nearly double the capacity (rated: 350mA). With a beacon every 5 seconds and normal TX, should last nearly 2 years in my calculations.

landrysik, 2024-02-12:
As a replacement for CR2032 for modified zigbee buttons and sensors in different housing I use CR123. They have 1.5Ah (compared to 200mAh of CR2032). The chemistry is exactly the same as CR2032.
 
Another batteries I like to use are Energizer Lithium AA/AAAs. They are slightly different chemistry (no load voltage is around 1.7-1.8V/cell) so if you use 2of them, you get close to the desired voltage (3.6V is still moreless within operating parameters of the 3V stuff).  AAAs have around 1200mAh and AAs have around 3000mAh of capacity and really long shelf life. So you can achieve theoretical battery lifetime of 15years on two lithium AAs! (assuming on CR2032@200mAh it lasts 1year).

Here is a modifier zigbee switch (aqara) where the actual microswitch is desoldered and wall switch button contact is soldered instead. The advantage is seamless design of other switches and wall sockets in house and possibility to use the wireless zigbee switch in the same bracket as wall socket. And while I was at it, I replaced the CR2032 for two lithium AAAs.
[image]

Yes the batteries are expensive but does not really matter when the lifetime is like 6years...
PS: if you want to use rechargeable 18650, go for LiFePO4. The voltage looks ideal. MAximum charging voltage is 3.6V and minimum cutoff is around 2.5V

ilgrank, 2024-02-15:
thanks for your comment @landrysik! 
I had already in mind to use a pair of lithium AAAs, as it seems to be the only chemistry that handles well temperatures below -20°C.
As @pvvx demonstrated tho, even cheap Alkalines seems to fare well 'enough' 
LiFePO4 in the contrary, doesn't like low temps at all
Using CR123A seems to be a nice idea: they're compact, and even accounting for low temperature they could retain enough capacity to feed a sensor for 2 to 3 years

guillermohor, 2024-02-15:
I'm using a lithium 2032 cell and I was a 4.2v full charge.. it worked for more than 3 months without. Problems. I have charged it again and still ok.

ilgrank, 2024-02-16:
@guillermohor : a CR2032 with 4.2 voltage? Are you sure? a LIR2032 should also be 3.6V, not 4.2..

maxoss, 2024-02-16:
Have the same, fully charged LIR2032 is around 4.07V

ilgrank, 2024-02-16:
@maxoss @guillermohor : thanks both for your report.
According to what pvvx experienced, now your sensors could be consuming more than expected tho.

slingel, 2024-02-22:
Do you guys know about the Holtek 7533 ?
Input voltage up to 30V, quiescent current max 4µA.

[image]

landrysik, 2024-02-22:
@slingel That is a good tip, I use them aswell. Just be sure to get genuine ones. Fake copies can have much worse performance (higher quiescent current, lower maximum current and worse voltage drop). 
This dude tested originals VS clones : https://chiptron.cz/articles.php?article_id=267
Toss this page to google translate, it is in Czech. Or just look at graphs

slingel, 2024-02-25:
@landrysik
Thanks for the link.
I tested my 7533 clone with a 56 Ohm load connected to the output.
Voltage dropped from 3,32V to 3,2V at ~85% efficiency.
Voltage spikes above 3,5V when load is connected.
It idles at 1,8µA.
7833 I don't have.

ilgrank, 2024-02-26:
> @landrysik Thanks for the link. I tested my 7533 clone with a 56 Ohm load connected to the output. Voltage dropped from 3,32V to 3,2V at ~85% efficiency. Voltage spikes above 3,5V when load is connected. It idles at 1,8µA. 7833 I don't have.
@landrysik : isn't a linear regulator more efficient with such low deltas? It would be 96% efficient, with near zero quiescent current in this scenario..

landrysik, 2024-02-26:
@ilgrank you don't really calculate efficiency for linear regulators. Of course the best "efficiency" you would get with 3.1V input and 3.0V output...but where do you get 3.1V input? You must work with what you get based on the chemistry/SOC of the battery etc...so if the battery has 4V and you need 3V, you "burn to heat" 1V multiplied by the current you draw. For higher voltage differences (from 5V or even from 12V down to 3V)  you need to get switching step-downs and then you can calculate efficiency. But they usually get higher quiescent currents compared to LDOs. You must use what is better suited for your application and current draws. For this it is best to get LDO with lowest quiescent current and low dropout. So it stays in regulation (3.0V output) even when the li-ion battery is nearly discharged at 3.3V or so. On the other hand it does not matter much in this case as these thermometers work down to 2V or so

ilgrank, 2024-02-26:
@landrysik : I know how a voltage regulator works, but thanks anyway :) 
I totally agree with you on the general term, in fact I opened this topic just about using a switching regulator, as you can see in the first picture. But since you wrote you had 3.2V, i suggested  you that in that specific scenario (3.3V regulated to 3.2V) , a linear regulator would have been more efficient

pvvx, 2024-02-26:
Also remember that higher voltage produces higher leakage current. Relevant for the period of processor sleep and operation of the LCD controller.
As a result, the drop across the LDO or even the diode offsets the resulting current a little...

ilgrank, 2024-02-27:
Thanks for the data @pvvx !
Btw, may I ask you what instrument are you using for the measures?

pvvx, 2024-02-27:
[nRF Power Profiler kit II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2)
Doesn't work well with dynamic signals.
Normal measurements at low signal amplitude, if the range switching threshold does not cross...
The automatic range switching circuit introduces large distortions into the displayed signal...
[image]

auto-switch 1
[image]

auto-switch 2
[image]

auto-switch 3
[image]

...

NRF did it - what can you do :)

ilgrank, 2024-02-28:
Wow, thanks @pvvx , I did not know about your power profiler!

AJolly, 2026-03-24:
FYI, With an RC123A  so 4.20v it works but it overvolts the lcd so it makes it much harder to read.

DiegoJp, 2026-09-30:
2nd FYI, Using a rechargeable LIR2032 (4.2 max nominal) also overcooks the chip, LCD will contrast more and humidity sensor will offset to +50%