pvvx/ATC_MiThermometer #480: Rogue bluetooth connections

koisama, 2024-02-12:
I have a bunch of LYWSD03MMC's flashed with custom firmware. However it seems like some rogue device is constantly attempting to connect to one of those over bluetooth, and once it succeeds, it keeps the connection open forever, draining the battery and preventing the sensor from broadcasting the data. I confirmed this by taking the sensor outside and observing that bluetooth icon would no longer show up on the display.

I would like to secure the sensors. Is there any possibility to ignore unwanted connections?

pvvx, 2024-02-12:
Set a PIN code. Remember your PIN code, because... There is no way to reset it if you forget.

fanoush, 2024-02-15:
some mac address white list in thermometer could help with this, it would not allow connection or  immediately disconnect from devices with address not in the list. too bad there is no button to override it or enable some 'learning mode' that would add currently connecting device to the whitelist when button is held.

pvvx, 2024-03-09:
The pin code is made in such a way that it is impossible to change anything. Removing the Pin code is only possible by flashing the hardware with eraser of the entire Flash.