pvvx/ATC_MiThermometer #482: Arduino ESP32 library

ioukos, 2024-02-25:
Hi guys,

https://github.com/matthias-bs/ATC_MiThermometer

Is this Arduino library compliant with sensor reading using [LYWSD03MMC] flashed with your firmware ? 
I mean for temperature and humidity. May be for reed switch reading I will need some changes in the arduino library.

If no, is there a compliant library I can use ?

Thank you for your amazing work.

chinswain, 2024-04-02:
I use this: https://github.com/matthias-bs/ATC_MiThermometer
Does not have the reed switch though, just Temperature, Humidity, Battery and signal strength.

chinswain, 2024-04-03:
The library owner has responded so there is hope.
I've also started to try and modify the library to include the reed switch count and flags of the custom advertisement. Feel free to assist if you can.

https://github.com/chinswain/ATC_MiThermometer

Currently the Reed Switch State is working (open\closed), I cannot work out how to extract the count though position [13] appears to increment on each advertisement so not the reed count.

Sensor 1: a4:c1:38:9f:03:47
23.70°C
46.49%
2.931V
100%
-57dBm
Count: 73
Reed Switch State: Closed
GPIO TRG Output: Low
Control Parameters: Set
Temperature Trigger Event: Not Triggered
Humidity Trigger Event: Not Triggered

chinswain, 2024-04-29:
Hi @ioukos
The owner has merged my pull request - could you test the latest version of the library?
I can't work out how to get the Reed Switch count though, it's shown on the flashing tool\website so must be contained in the message somewhere.

Usage:
```
Serial.printf("Measurement Count: %d\n", miThermometer.data[i].count);
// Print boolean flags
Serial.printf("Reed Switch State: %s\n", miThermometer.data[i].reedSwitchState ? "Open" : "Closed");
Serial.printf("GPIO TRG Output: %s\n", miThermometer.data[i].gpioTrgOutput ? "High" : "Low");
Serial.printf("Control Parameters: %s\n", miThermometer.data[i].controlParameters ? "Set" : "Not Set");
Serial.printf("Temperature Trigger Event: %s\n", miThermometer.data[i].tempTriggerEvent ? "Triggered" : "Not Triggered");
Serial.printf("Humidity Trigger Event: %s\n", miThermometer.data[i].humiTriggerEvent ? "Triggered" : "Not Triggered");
```

I'm not sure if my interpretation of the custom format is correct though so please check and correct.
```
// Count
data[n].count = foundDevices.getDevice(i).getServiceData().c_str()[13];    
                  
//Flags
uint8_t flagsByte = foundDevices.getDevice(i).getServiceData().c_str()[14];
data[n].reedSwitchState = flagsByte & 0x01; // Extract bit 0 (Reed Switch)
data[n].gpioTrgOutput = (flagsByte >> 1) & 0x01; // Extract bit 1 (GPIO_TRG pin output)
data[n].controlParameters = (flagsByte >> 2) & 0x01; // Extract bit 2 (Control parameters)
data[n].tempTriggerEvent = (flagsByte >> 3) & 0x01; // Extract bit 3 (Temperature trigger event)
data[n].humiTriggerEvent = (flagsByte >> 4) & 0x01; // Extract bit 4 (Humidity trigger event)
```

Custom format:
```
uint8_t     size;   // = 18
uint8_t     uid;    // = 0x16, 16-bit UUID
uint16_t    UUID;   // = 0x181A, GATT Service 0x181A Environmental Sensing
uint8_t     MAC[6]; // [0] - lo, .. [6] - hi digits
int16_t     temperature;    // x 0.01 degree
uint16_t    humidity;       // x 0.01 %
uint16_t    battery_mv;     // mV
uint8_t     battery_level;  // 0..100 %
uint8_t     counter;        // measurement count
uint8_t     flags;  // GPIO_TRG pin (marking "reset" on circuit board) flags: 
                    // bit0: Reed Switch, input
                    // bit1: GPIO_TRG pin output value (pull Up/Down)
                    // bit2: Output GPIO_TRG pin is controlled according to the set parameters
                    // bit3: Temperature trigger event
                    // bit4: Humidity trigger event
```

ioukos, 2024-05-06:
Thank you for the update.
Will try ASAP.

Just to be sure, this library is always "passive", I mean just "listenning" and not polling the devices ?

Thank you

chinswain, 2024-05-06:
Correct, it captures the advertisement of known devices, no connection.