pvvx/ATC_MiThermometer #484: One of my reports 0C temp and 0% humidity after upgrade to v4.7

m-kozlowski, 2024-02-29:
Downgrading to v4.6 brings it back to normal operation.

Identification string it sends on v4.7:
`Hardware Version: LYWSD03MMC B1.4, Software Version: 4.7, Sensor: SHTC3 (SHTV3)`
vs v4.6:
`Hardware Version: LYWSD03MMC B1.7, Software Version: 4.6, Sensor: SHT4x`

board:
CHMLmini_TLW_v2.0
202107
HW: B2.0
[image]

dspalu32, 2024-02-29:
Same thing for me.

Worked OK on v4.6, 0.00 values on v4.7 (both on _TelinkMiFlasher.html_ and in Home Assistant), and working OK again after moving back to v4.6

Thanks for this firmware and all the improvements

pvvx, 2024-03-01:
Ver4.7 is still a Beta version. Not tested on all thermometer options.

vartom, 2024-03-09:
Установив прошивку 4.7 бета получил удвоение значения температуры. 
[image]
[image]
Вернув 4.6 всё нормализовалось.

pvvx, 2024-03-09:
Коэффициенты для датчика исправлены.
[image]

enrico-pallazzo, 2024-04-21:
How can i change humidity / temperature setting on v4.7?
On 4.6 it was just temperature and humidity offset.
Need to set temperature offset 0.5 deg and humidity -4%

pvvx, 2024-04-21:
Add offset to "Zero offset".

satmandu, 2024-10-26:
I'm also seeing 0 degrees C, 0% humidity when updating to the Z03MMC_v0123.bin zigbee firmware for the LYWSD03MMC B1.7 hardware. Is there an alternate zigbee firmware I should try for this?

closed 2025-04-23 by pvvx