pvvx/ATC_MiThermometer #485: Temperature reading instability 4.3/4.4 vs 4.6 and SHT4x/SHTC3

apppie123, 2024-03-04:
I have 4 sensors here, see below with their config(it's all the same, except for adflag and BT5+ PHY difference).

Sensor 1 and sensor 2 give very stable readings.
However, readings from sensor 3 and sensor 4 jump all over the place on the display.
They do this for example: 18.9 -> 19.7 -> 19.2 -> 19.4 -> 19.1 -> 18.9 -> 19.3

I noticed #3 and #4 have an SHTC3 sensor while #1 and #2 have a SHT4x sensor.

Anyone else noticed this? Should I use different settings?

```
Sensor 1
HW B1.6
SW V4.3
Sensor SHT4x
config 55a42100002804bf7c3104b4

Sensor 2
HW B1.6
SW V4.4
Sensor SHT4x
config 55a41000002804bf313104b4

Sensor 3
HW B1.5
SW V4.6
Sensor SHTC3
config 55a40000002804bf31318ab4

Sensor 4
HW B1.5
SW 4.6
Sensor SHTC3
config 55a40000002804bf31318ab4
```

pvvx, 2024-03-04:
SHTC3 in
[image]

apppie123, 2024-03-04:
Thank you very much for such a quick reply! I was just going to ask why there is a difference, but you edited the message already.

Thanks a lot. I'm going to try it right away.

pvvx, 2024-03-04:
For SHT4x, “lowPower” is enabled by default (at first startup after the first firmware).

closed 2024-03-09 by pvvx