pvvx/ATC_MiThermometer #486: Advertisements incidentally fail decryption

rrooggiieerr, 2024-03-05:
Advertisements incidentally fail decryption. I'm using a _LYWSD03MMC B1.6_ and the `ATC_v46.bin` firmware

Hereby 3 advertisements where decryption failed. Bind key: `ea1327b1f733cfd2e0deeb6304d9e286`

```
2024-03-05 01:25:29.518 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (B8:27:EB:B0:2C:A1) [connectable]: A4:C1:38:8F:19:E9 AdvertisementData(local_name='Kitchen', service_data={'0000fcd2-0000-1000-8000-00805f9b34fb': b'A<}\xba\xf1|\xe1\x18\x00\x8f*\x0c\x00\x1e\xf5\x13\xfd'}, service_uuids=['0000fcd2-0000-1000-8000-00805f9b34fb'], rssi=-79) match: set()
2024-03-05 01:25:29.519 DEBUG (MainThread) [bthome_ble.parser] Parsing BTHome BLE advertisement data: <habluetooth.models.BluetoothServiceInfoBleak object at 0x7f58ba5140>
2024-03-05 01:25:29.575 WARNING (MainThread) [bthome_ble.parser] Decryption failed: 
2024-03-05 01:25:29.582 DEBUG (MainThread) [bthome_ble.parser] mic: 1ef513fd
2024-03-05 01:25:29.583 DEBUG (MainThread) [bthome_ble.parser] nonce: a4c1388f19e9d2fc418f2a0c00
2024-03-05 01:25:29.583 DEBUG (MainThread) [bthome_ble.parser] encrypted_payload: 3c7dbaf17ce11800
```

```
2024-03-05 06:15:51.869 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (B8:27:EB:B0:2C:A1) [connectable]: A4:C1:38:8F:19:E9 AdvertisementData(local_name='Kitchen', service_data={'0000fcd2-0000-1000-8000-00805f9b34fb': b'A\x173\xa4\x17l}\xc1-\xae\xa7\x00\xdb\x07i\xb3\x04'}, service_uuids=['0000fcd2-0000-1000-8000-00805f9b34fb'], rssi=62) match: set()
2024-03-05 06:15:51.870 DEBUG (MainThread) [bthome_ble.parser] Parsing BTHome BLE advertisement data: <habluetooth.models.BluetoothServiceInfoBleak object at 0x7f518b2640>
2024-03-05 06:15:51.877 WARNING (MainThread) [bthome_ble.parser] Decryption failed: 
2024-03-05 06:15:51.878 DEBUG (MainThread) [bthome_ble.parser] mic: 0769b304
2024-03-05 06:15:51.878 DEBUG (MainThread) [bthome_ble.parser] nonce: a4c1388f19e9d2fc41aea700db
2024-03-05 06:15:51.879 DEBUG (MainThread) [bthome_ble.parser] encrypted_payload: 1733a4176c7dc12d
```

```
2024-03-05 09:36:50.536 DEBUG (MainThread) [homeassistant.components.bluetooth.manager] hci0 (B8:27:EB:B0:2C:A1) [connectable]: A4:C1:38:8F:19:E9 AdvertisementData(local_name='Kitchen', service_data={'0000fcd2-0000-1000-8000-00805f9b34fb': b'A$\xf6\xc4\x8c\xa3\xfe\xedv\x14?\x0c\x00\x8d\x04\x04\x00'}, service_uuids=['0000fcd2-0000-1000-8000-00805f9b34fb'], rssi=62) match: set()
2024-03-05 09:36:50.537 DEBUG (MainThread) [bthome_ble.parser] Parsing BTHome BLE advertisement data: <habluetooth.models.BluetoothServiceInfoBleak object at 0x7f5fd254c0>
2024-03-05 09:36:50.541 WARNING (MainThread) [bthome_ble.parser] Decryption failed: 
2024-03-05 09:36:50.542 DEBUG (MainThread) [bthome_ble.parser] mic: 8d040400
2024-03-05 09:36:50.543 DEBUG (MainThread) [bthome_ble.parser] nonce: a4c1388f19e9d2fc41143f0c00
2024-03-05 09:36:50.543 DEBUG (MainThread) [bthome_ble.parser] encrypted_payload: 24f6c48ca3feed76
```

Also see https://github.com/Bluetooth-Devices/bthome-ble/issues/107

pvvx, 2024-03-05:
Advertisement data is corrupted.
What BT adapter are you using?

rrooggiieerr, 2024-03-05:
Raspberry 3 onboard BT

pvvx, 2024-03-05:
The BT adapter in Raspberry 3 often makes errors when transmitting via UART.

rrooggiieerr, 2024-03-06:
Close as it is a hardware problem

closed 2024-03-06 by rrooggiieerr