pvvx/ATC_MiThermometer #488: No bluetooth After reverting from zigbee to BLE

tamamma, 2024-03-09:
I've used the tool for installating zigbee firmware. Then I had to go back to BLE and I've tried installing through UART tool both the original firmware and the ATCv46 and v47 but I can't see anymore the device on Bluetooth neither with the Xiaomi home app than the telinkflasher web page :(

pvvx, 2024-03-09:
Why did you use UART for flashing?

https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota#readme

---

You need to flash the file:
https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v47.bin

README: [The USB-COM adapter writes the firmware in explorer. Web version](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)
In difficult cases, when the chip contains an unknown FW, it is necessary to use a chip reset or power supply during "activation".

tamamma, 2024-03-10:
I've used USB-COM because I couldn't use the zigbee. A friend of mine should give me his dongle but he sold it :(. Flashing with the usbcom didn't give me errors in the log so I think is correctly flashed but I cant see it on Bluetooth.

landrysik, 2024-03-17:
> ... Flashing with the usbcom didn't give me errors in the log so I think is correctly flashed but I cant see it on Bluetooth.

Be aware that the USB-COM flasher has no feedback! So it will give "OK" even if there is no microcontroller connected!

lardeg, 2024-03-19:
@pvvx 
Как должен происходить откат с Zigbee на BLE?
Прошил через ОТА [1141-020a-99993001-ATC_v46.zigbee] датчик не перешел в режим BLE и всё еще отображается в z2m

pvvx, 2024-03-19:
Значит не OTA в Z2M не отработало.

closed 2025-04-23 by pvvx