pvvx/ATC_MiThermometer #489: No screen (LYWSD03MMC 1.4 bricked ?) after downgrade tentative

yan-064, 2024-03-09:
Hello,

I request your help after reading a lot of issues. For the context, I tried to downgrade a LYWSD03MMC v1.4 from 2.1.1_0159 to 1.0.0_0130 and now it seems to be bricked. No screen, no bluetooth connection.

For information, I use a FT232R usb-com adapter on a Windows computer. I read that Mac users seems to have the same trouble. I tried also on a second Windows computer, same result.
 However, I flashed with the online [tool](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) that returned success state many times with 2 or 3 wires as described [here](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version) or [here](https://github.com/atc1441/ATC_MiThermometer/issues/298).

Anyone could please help me ? I don't know what to do.

Thanks.

pvvx, 2024-03-09:
You need to flash the file:
https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v47.bin

README: [The USB-COM adapter writes the firmware in explorer. Web version](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)
_In difficult cases, when the chip contains an unknown FW, it is necessary to use a chip reset or power supply during "activation"._

yan-064, 2024-03-09:
Thanks for your feedback.

Yes, I saw that section "in difficult cases, when the chip contains an
unknown FW, it is necessary to use a chip reset or power supply during
"activation",  but I don't understand what I really have to do.

I tried to put TX on vbat whithin the atime period and then put it back on
p14. No effect, I don't know if it is the good handling.

I'll try with the atc_v47 instead of the xiaomi_ota_1.0.0.109.

Thanks

yan-064, 2024-03-10:
Hi, I just tried with the atc_v47, always the same result. No screen, no bluetooth. It seems to be dead.

Who could explain me how to do that please ? => _"In difficult cases, when the chip contains an unknown FW, it is necessary to use a chip reset or power supply during "activation"_

I ever read that _"At the same time, take into account that the chip can be powered from the TX output. This is solved by connecting TX and +Vbat during "activation"_. But I don't understand what I exactly have to do.

Does that mean that I have to :
- plug gnd to gnd
- plug Tx to vbat+ 
- click write to flash
- before 3s ellapsed, unplugged the Tx from vbat+ and plug it to p14
- wait for flash 

I'm right ?

Thanks.

pvvx, 2024-03-10:
[image]

1. Remove battery
2. GND -> GND
3. Set "Atime" to maximum value
4. "Erase All Flash"  
5.  Quickly connect: TX -> P14, 3V3 -> +Vbat

After these steps, when connecting the power, you will not need to reset or connect pins on the fly and programming should be successful.

yan-064, 2024-03-10:
Thanks again for your help. Unfortunately, same result. I follow your steps like that :

- remove battery
- plug gnd on gnd
- set atime to 16s
- click erase all flash => countdown started
- quickly connect (before 16s) tx on p14 and 3.3v on vbat+
- wait for erase => success

Then I try to flash with atc_v47.bin.

- still no battery
- plug gnd on gnd
- plug tx on p14
- set atime to 3s and let default bauds (46080)
- select atc_v47 firmware
- click "Write to flash" ==> success

Logs here :
21:02:36: USB-COM opened.
21:02:38: File was selected, size: 83028 bytes
21:03:13: Reset DTR/RTS (100 ms)
21:03:13: Activate (3 sec)...
21:03:16: Write 83028 bytes in to Flash...
21:03:47: Done (34.521 sec).
21:03:47: Soft Reset MCU

Still not working, that makes me mad. Why did I update that Xiaomi firmware before checking on the web, so stupid.

zawatowskib, 2024-04-23:
@yan-064 did you get any chance solving this? I too could not get it to work.
I assume the instructions were translated using google translate, so they are a bit vague.
I have a FTDI based USB COM, might this be the issue?

pvvx, 2024-04-23:
Option 2:

[image]

https://github.com/pvvx/TlsrComSwireWriter

File "https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py"

There is an example here:
[Flashing firmware with USB to UART](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-firmware-with-usb-to-uart)

yan-064, 2024-04-25:
> @yan-064 did you get any chance solving this? I too could not get it to work. I assume the instructions were translated using google translate, so they are a bit vague. I have a FTDI based USB COM, might this be the issue?

Hello, no I gave up. I don't know why it's not working for me. 
I'll look at the option 2.

zawatowskib, 2024-04-25:
> Option 2: [...]

Thanks, will try this later.
I'll need another UART converter, as FTDI chip is indeed not supported apparently.

wlfbck, 2024-11-26:
> ![image](https://private-user-images.githubusercontent.com/12629515/311516141-8120c146-2e02-44de-97c9-d8262d65ae21.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I have the CH340C adapter. When i try to do this, as soon as i connect 3V3 to the +Vbat the USB will disconnect. Any idea how else i could reset a blank device?

pvvx, 2024-11-26:
> I have the CH340C adapter. When i try to do this, as soon as i connect 3V3 to the +Vbat the USB will disconnect. Any idea how else i could reset a blank device?

This means that the 3.3V power supply is short-circuited. Something is short-circuited or the chip in the thermometer has already been burned out...

wlfbck, 2024-11-27:
> This means that the 3.3V power supply is short-circuited. Something is short-circuited or the chip in the thermometer has already been burned out...

That's what i feared. I had to hold down the connector on P14 when flashing with USB with my hand, i must have accidentally shorted something while doing that. I only ever used 3V3, GND and TXT though.

Just to clarify, in your picture you also have a green wire to the reset pin? That is not mentioned in the steps?

wlfbck, 2024-11-27:
> > I only ever used 3V3, GND and TXD though.
> 
> Check for connection errors. Nothing will burn out from 3.3V, unless you mix up the positive and negative power supplies and leave it on for a long time.

Found the issue. It was faked cables from china which were made of steel not copper -> high resistance. https://www.youtube.com/watch?v=15sMogK3vTI

As soon as i switched from that to jumper wires, actually made out of copper, it worked in the first try.

So yeah, connection error indeed!

closed 2025-04-23 by pvvx