pvvx/ATC_MiThermometer #491: Request for Documentation on Managing Time Settings via Bluetooth Commands

AlexMihai1804, 2024-03-16:
I'm utilizing the ATC_MiThermometer firmware and greatly appreciate its capabilities. However, I've found it challenging to operate without comprehensive documentation, particularly regarding setting and managing the device's time via Bluetooth commands, such as using command ID 0x23 for "Get/Set Time".

Could detailed documentation or examples be provided on how to format and send these time-related commands? Information on byte structure and examples for setting the time would be immensely helpful for developers to effectively use this firmware.

pvvx, 2024-03-23:
PrimaryService UUID16 0x1F10
Characteristic UUID16 0x1F1F

Enable Notifications.

CMD ID: 0x23 - Get/set utc time

Send/receive:
```
// set time
typedef struct {
	uint8_t		id;	// = 0x23 
	uint32_t	time; // Uinux timestamp (https://www.unixtimestamp.com/)
} cmd_set_time_t

// get time
typedef struct {
	uint8_t		id;	// = 0x23 
} cmd_get_time_t

// Reply to command
typedef struct {
	uint8_t		id;	// = 0x23
	uint32_t	current time; // Uinux timestamp (https://www.unixtimestamp.com/)
	uint32_t	old_set_time; // When was the last time there was a command to set the time, Uinux timestamp
} msg_memo_t
```

CMD ID: 0x24 - Get/set adjust time clock delta (in 1/16 us for 1 sec)

adjust_time default = 0

Send/receive:
```
// set adjust time
typedef struct {
	uint8_t		id;	// = 0x24 
	int16_t	adj_time; // in 1/16 us for 1 sec -> step_time = 16000000 + adj_time
} cmd_set_adj_time_t

// get adjust time
typedef struct {
	uint8_t		id;	// = 0x24 
} cmd_get_adj_time_t

// Reply to command
typedef struct {
	uint8_t		id;	// = 0x24
	uint32_t	step_time; // in 1/16 for 1 sec (default 16000000)
} msg_memo_t

```

pvvx, 2024-03-23:
If a real RTC clock is installed, then there is no clock adjustment.
The answer to command `24` is `24FF`.
Response to command `23`:
```
// Reply to command 0x23
typedef struct {
	uint8_t		id;	// = 0x23
	uint32_t	current time; // Uinux timestamp (https://www.unixtimestamp.com/)
} msg_time_t
```

PS: To find out which commands and services are used by firmware from version 4.6, perhaps by command 0x00.

Ircama, 2024-03-25:
Check also [atc_mi_config](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface#atc_mi_config-configuration-tool-with-gui). The full message formats supported by the firmware are described in [atc_mi_construct.py](https://github.com/pvvx/ATC_MiThermometer/blob/master/python-interface/atc_mi_interface/atc_mi_construct.py) and [atc_mi_config.py](https://github.com/pvvx/ATC_MiThermometer/blob/master/python-interface/atc_mi_interface/atc_mi_config.py#L33-L34). Examples [of usage](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface#examples-of-usage).