pvvx/ATC_MiThermometer #492: MJWSD05MMC only shows Temp: 0.00°C, Humi: 0.00%

sorenbache, 2024-03-16:
Hi,

I have a MJWSD05MMC that I (finally - it was very very hard to get activated, pure luck it seems) flashed with 4.7, but it really only shows 0.00°C, Humi: 0.00% as shown here after connection:

[image]

Any input to what could be going on here? 

PS. I have another MJWSD05MMC (and a bunch of LYWSD03MMC) that works fine connected to home assistant.

pvvx, 2024-03-16:
There have never been any problems with the MJWSD05MMC.
Why did You flash the firmware from LYWSD03MMC in MJWSD05MMC?

sorenbache, 2024-03-17:
Hmm...I wonder how that could happen? Where can I find the firmware for MJWSD05MMC for manual download - then I can try to flash that one.

pvvx, 2024-03-17:
https://github.com/pvvx/ATC_MiThermometer/raw/master/BTH_v47.bin

Didn't [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) offered this firmware after activation?

sorenbache, 2024-03-17:
So I have not done anything manual - only used the Telink flasher. I cannot remember why it might have flashed the wrong LYWSD03MMC firmware, but as I said - nothing manual from my side - hence, I can only explain it as the flasher must somehow have presented me with the wrong firmware.

Anyway, I downgraded the FW to the original firmware and managed (after a few attempts) to activate and flash the BTH_v47.bin firmware.

closed 2025-04-23 by pvvx