pvvx/ATC_MiThermometer #495: Про чтение из LYWSD03MMC сохраненых результатов

JurySpb, 2024-03-20:
Есть ли какая то доп. информация? Знаю, что можно читать с помощью Веб-интерфеса. Есть ли другие способы? Хотелось бы с помощью Linux утилит, например **gatttool** и python.

JurySpb, 2024-03-20:
Подробней можно? Уточняю, меня интересуют сохраненные во флэш памяти где то 20000 отсчетов за длительное время.  Получение методами bluetoth LE.

KirillMMM, 2024-03-20:
Не там только онлайн считивания показаний

JurySpb, 2024-03-20:
Ну как же, вот на гл. странице написано:
_Reading Measurements from Flash

[GraphMemo.html](https://pvvx.github.io/ATC_MiThermometer/GraphMemo.html)

To record the measurement results, a cyclic buffer in the flash memory for 20000 measurements is used. The recording step interval is set in the interface. With the default settings, the recording step is 10 minutes, which gives a recording depth of 4 and a half months. During the step period, the sensor data and battery voltage are averaged, time stamped, and written to flash memory._

pvvx, 2024-03-23:
https://github.com/pvvx/ATC_MiThermometer/discussions/440#discussioncomment-8885144