pvvx/ATC_MiThermometer #498: Telink Flasher v7.7 connecting to LYWSD03MMC crashes Chrome on Android 14

ievren, 2024-03-28:
Hi @pvvx thank you for your great work and optimisations and documentation.

Tried to use Android phone and connecting to a LYWSD03MMC will crash my chrome browser(Version 123.0.6312.41) on Android 14. I also tried to use vivaldi with the same behaviour that as soon the Bluetooth logo appears on the LYW... the browser on the phone crashes...

I just noticed that when I try to connect with a laptop chrome (ubuntu) I managed to connect to it once and flashed it with the atc firmware:
Status: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.6

Then I lost the connection tried to connect with the laptop again using chrome I was not able to connect.(So I assume my laptop has some issues)

I tried with another phone using android 13 (chrome browser(Version 123.0.6312.41)) and I was able to connect :)

Status: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.3
Vbat: 2985 mV , Temp: 24.27°C, Humi: 55.84%, ID: 14, Reed switch counter: 0, flg: 0x04:r0/t0

So this is probably more a android bug using BLE connection, maybe? But maybe someone will have same issue...
[image]

pvvx, 2024-03-28:
> Chrome (ubuntu) 

README:  
_5. On Linux, when installing large advertising intervals, change the file '/etc/bluetooth/main.conf', variables: LEScanIntervalAutoConnect= LEScanWindowAutoConnect=. Value in ms._

ievren, 2024-03-28:
Thank you. 

Yes adding `/etc/bluetooth/main.conf` works now to connect with the linux laptop.
```
LEScanIntervalAutoConnect=2500 
LEScanWindowAutoConnect=1000
```
I am not able to use it with android 14 + chrome ( 123.0.6312.41) ...
But there are these workaround which are ok for me laptop or android 13 are working perfectly :)

Maybe #459 is duplicate of this. But @Robert-Versteegen is using win11. And no info about which android...

closed 2024-03-28 by ievren

pvvx, 2024-03-29:
> LEScanIntervalAutoConnect=2500  LEScanWindowAutoConnect=1000

To comply with the Bluetooth specification, these parameters must use a period of up to 10 seconds. But if there is a loss of reception (collisions on the radio), then the value will be increased by 2. That is, from 20 seconds.