pvvx/ATC_MiThermometer #499: How connect to HomeAssistant after flashing BTH_v47.bin to MJWSD05MMC?

zhongqitech, 2024-03-29:
Hi all,

I had flashed the BTH_v47.bin firmware to MJWSD05MMC with BTHome v2 advertising type. But can't do that when connecting to HomeAssistant by ['BTHome integration'] & ['Passive BLE Monitor']. The strange thing is that the Bluetooth device is never detected on HomeAssistant.

Could anyone help me out or have a detailed operation manual for the BTH_v47.bin firmware?

pvvx, 2024-03-29:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

[image]

zhongqitech, 2024-03-30:
@pvvx

After again, That's it! You helped me get out of my confusion for many days. Thank you!