pvvx/ATC_MiThermometer #500: Upgrade issue: No battery stats in home assistant after firmware version 4.3

Zephael86, 2024-04-06:
Since firmware 4.3 and up I always have 100% battery in Home Assistant.

Story time:
I have multiple Xiaomi Mijia at home. Hardware revision B1.4.
Data is collected with the help of a gateway ESP32, programmed via ESPHome.

I am using them since 2021, nearly the beginning of the first pvvx fork.
Firmware update on the thermometers happens, whenever I replace the battery.
(Therefore I still have 3.7, 4.2, 4.3 and 4.7 at the moment.)
Then I open the Telink Flasher, connect to the device and upload the latest available firmware.
For a year now I do not even note down the firmware version, because the flasher is so convenient.

I did not realized this problem only after 6/10 devices always reports 100% battery.
Temperature and Humidity seems to work fine.

Example ESPHome config:
```
- platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:18:51:7D"
    #Dummy bindkey
    bindkey: "eef418daf699a0c188f3bfd17e4565d9"
    temperature:
      name: "Bathroom temperature"
    humidity:
      name: "Bathroom humidity"
    battery_level:
      name: "Bathroom battery"
```

Advertising Type on device: MIJIA (MIHOME)

I would also appreciate if you could compile an ATC_42.bin, so I can retest if the now 4.3 and up devices start to send battery info.
I tried to setup a build environment, but was not successful so far.

pvvx, 2024-04-06:
This thermometer used all versions.

[image]

[image]

Works in НA and simultaneously in Mi-Home via Xiaomi Gateway 3.

Zephael86, 2024-04-06:
Any idea where to start looking for errors?

pvvx, 2024-04-07:
Xiaomi BLE integration into Home Assistant successfully shows all values:
[image]

It is also worth checking that the “Measurement interval” value is set to at least 3 (relevant for the Xiaomi-Mi-Home format).
[image]

This means that the errors are not in the thermometer or in Home Assistant, but in your gateway or BLE receiver.

Zephael86, 2024-04-21:
I could play with the thermostats at the weekend.
[Advertising](https://pvvx.github.io/ATC_MiThermometer/Advertising.html) and a mobile phone also showed 100% battery for me. Therefore I suspect that something went wrong during upgrades.

I did not change neither my ESP32 gateway nor any Home Assistant config.

I flashed back first `ATC_v42.bin` then `ATC_v37d.bin`.
First was not a success, but after the second flash I started to receive battery percentage data other than 100%.

I believe that with Xiaomi BLE integration it works. I already have my separate ESP32 Gateway deployed and cannot switch to a dongle, because my HA server runs in the basement.

_From my point of view the issue is fixed with a workaround.
In case you want, I can verify a thing or two for you, in case somebody else is still using ESP32 gateways._

pvvx, 2024-04-21:
https://github.com/pvvx/ATC_MiThermometer/issues/507#issuecomment-2067538991

avg-I, 2024-07-23:
I am having the same issue but without any involvement of Home Assistant.
Battery voltage is dropping, but battery level is always 100%.

E.g., right now the voltage is reported as 2950 mV, but the level is 100%.

Hardware version is LYWSD03MMC B1.4.
I am using advertising type ATC1441.