pvvx/ATC_MiThermometer #501: Update issue: no temperature offset setting available in Telink Flasher

Zephael86, 2024-04-06:
For Xiaomi Mijia pvvx firmware 3.7, 4.2, 4.3 I can set temperature offset in Telink Flasher.
For 4.6 this option is missing.

pvvx, 2024-04-06:
ver 4.7 - Changing sensor settings: Added slope factor, zero offset in 0.01 units.

Zephael86, 2024-04-06:
I have to apologize. Indeed it was one device, that got updated to 4.7.
I can see this section called 'Sensor Settings'.
Now I have to understand :)

Shall I close, or you close it usually?

closed 2024-04-07 by pvvx

traeu, 2024-04-14:
After updating to 4.7, all my LYWSD03MMC show the same sensor settings:

For SHTv3:
Temperature slope factor: 175.00, Zero Offset: -45.00
Humidity slope factor: 100.00, Zero Offset: 0.00

For SHT4x:
Temperature slope factor: 175.00, Zero Offset: -45.00
Humidity slope factor: 125.00, Zero Offset: -6.00

Does that mean the calibration values I set before the firmware update are lost when updating to 4.7 and the sensors are now working with default values?

pvvx, 2024-04-14:
> Does that mean the calibration values I set before the firmware update are lost when updating to 4.7 and the sensors are now working with default values?

Yes, new values are used.

traeu, 2024-04-15:
> Yes, new values are used.

Maybe you could add a little notice to the changelog so people know that calibrations made in the past are not active anymore after updating to v4.7?

pvvx, 2024-04-15:
> Maybe you could add a little notice to the changelog so people know that calibrations made in the past are not active anymore after updating to v4.7?

[Firmware version history](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#firmware-version-history)
4.7 | Changing sensor settings: Added slope factor, zero offset in 0.01 units.

</body>
</html>

pvvx, 2024-04-17:
If the main settings have not been changed, then the offsets in them are saved.
When connecting in the log:
_Custom config HEX string: 55051000004004a9313180e1_

_1_2_3 _**4__5**
550510 **00 00** 04004a9313180e1

4th byte – offset for temperature of 0.1 C.
5th byte – offset for humidity of 0.1%.