pvvx/ATC_MiThermometer #502: [MISC] How to use a rechargeable battery

Toniob, 2024-04-07:
Hello,

This is not related to the firmware, but as there are a few experts of the LYWSD03MMC in here, I thought I could try. I've put a rechargeable battery in the sensor, a LIR2032. But this battery has a much higher voltage (3.7V). The sensor is working ok, but the screen is showing some kind of ghost image, so it is difficult to read.

Do you think there's a way to lower the voltage to make this kind of battery usable with a Mijia ?

JsBergbau, 2024-05-31:
Can you post a photo? Probably it is a too high contrast of the LCD which is optimized for lower voltages. Maybe you could/must change a SMD resistor to get proper voltage for the LCD again.

Toniob, 2024-05-31:
Here are the photos with a normal CR2032 : 
[image]
[image]

And with a LIR2032H : 
[image]
[image]

Toniob, 2024-06-02:
> That looks exactly as I supposed. Thats a normal behavior of a LCD getting too much voltage. Maybe it is possible to use another resistor for the LCD to get a normal contrast again. Have a look at this video https://www.youtube.com/watch?v=cF49mfYkyw0

Thank you for your answer. Do you know which one should be changed ?

pvvx, 2024-06-03:
1. It is not advisable to exceed the supply voltage more than 3.6V.
2. LYWSD03MMC were produced in 6 HW versions. LCD and controllers are different:

For LYWSD03MMC, the HW version is determined by the display and sensor addresses on I2C:
[README](https://github.com/pvvx/ATC_MiThermometer):
|HW | LCD I2C   addr | SHTxxx   I2C addr | Note |
|-- | -- | -- | -- |
|B1.4 | 0x3C | 0x70   (SHTC3) |  |
|B1.5 | UART | 0x70   (SHTC3) |  |
|B1.6 | UART | 0x44   (SHT4x) |  |
|B1.7 | 0x3C | 0x44   (SHT4x) | Test   original string HW |
|B1.9 | 0x3E | 0x44   (SHT4x) |  |
|B2.0 | 0x3C | 0x44   (SHT4x) | Test   original string HW |

Version 1.7 or 2.0 is determined at first run by reading the HW line written in Flash.
Display matrices or controllers are different for all versions, except B1.7 = B2.0.

Toniob, 2024-06-12:
Thank you for your answer. I mostly have 1.5 and 1.6 hardware versions. What kind of diode should I use ? How would you connect it ?

pvvx, 2024-06-12:
1N4148

Cossid, 2024-06-21:
I have HW B1.7 and the display appears to work fine with my LIR2032 batteries at 4.0V.  I did run into a different issue of the battery calculation overflowing, which I have submitted a PR ( #524 ) to correct.