pvvx/ATC_MiThermometer #504: Off topic: hacking TLSR8258 based PETKIT water fountain

wrobelda, 2024-04-11:
Hi there,

I have a Petkit Eversweet 3 Pro cat water fountain, which is heavily limited to only work via Bluetooth with the official PETKIT app. The main disadvantage is that you can't tell the water level if outside of its range, unless you have another PETKIT device that does both WiFi and Bluetooth and can work as a relay.

That having said, I opened it today to see what SoC it uses and its TLSR8258. Since this is same exact SoC as used by the Mija thermometers, I wonder how would one go about hacking that firmware?

Andurilll, 2024-05-04:
I am also interested in a custom Firmware. I have a eversweet solo 2 but did not open it yet. Should i? 

Note: this is a BLE device (not zigbee). Is These a Big difference in Firmware?

The board of OP reminds me a bit of a custom esp32?

euDominic, 2024-05-29:
I am also interested in a custom Firmware for the Petkit Eversweet 3 Pro

HVPereira, 2025-03-17:
How can we help to convert this to zigbee?

pvvx, 2025-03-17:
This is not a battery operated device or a thermometer.
For such devices, it is necessary to create a separate repository with its own support.

egormanga, 2025-03-17:
Is there anyone willing to provide some amount of assistance in implementing such firmware? The problem there is just lack of generic (not ATC-related) info on the chip or the toolkit.