pvvx/ATC_MiThermometer #506: Hardware version number corrupted on og firmware

HinataKato, 2024-04-16:
I have 2 LYWSD03 with hardware version 1.4. After upgrading to firmware 4.7 and then flashing back to the original firmware, the hardware version reported in both the Mi Home app and the Telink OTA webpage becomes corrupted data.
(Got another one with hardware version 1.4 that has been flashed with firmware 4.7, it should be the same, and hasn't been flashed back to the original firmware yet.)

[image]
[image]

pvvx, 2024-04-17:
Xiaomi firmware does not restore its data.
This doesn't interfere with anything.

HinataKato, 2024-04-17:
> Xiaomi firmware does not restore its data. This doesn't interfere with anything.

It does, this will cause TelinkMiFlasher not showing firmware to flash 
[image]

pvvx, 2024-04-17:
The new version of TelinkMiFlasher will load pointers to the firmware if there is no HW string