pvvx/ATC_MiThermometer #515: Using Bluetooth-Versions as Tracker for Googles Find my Device?

JsBergbau, 2024-05-31:
Google recently launched its find my device network. https://blog.google/products/android/android-find-my-device/
Basically the devices are bluetooth beacons. Is it possible to expand the firmware for the Mijia Bluetooth devices to use them as device for the Android Find My network? 

Since devices are still rare and currently sold quite expensive this would be a great alternative.

pvvx, 2024-06-01:
To do this, devices with a price of less than $1 are reprogrammed.

https://pvvx.github.io/iSearching/

aliexpress: Bluetooth 4.0 Gps Tracker ... 
https://nl.aliexpress.com/item/1005004397030802.html

It's worth contacting https://github.com/biemster/st17h66_FindMy

JsBergbau, 2024-06-01:
Thanks for your fast answer. 

> To do this, devices with a price of less than $1 are reprogrammed.

Sounds even better. 

> aliexpress: Bluetooth 4.0 Gps Tracker ...
> https://nl.aliexpress.com/item/1005004397030802.html

Sadly these aren't available in Germany. Aliexpress confusingly shows 
[image]

Aliexpress has a lot of cheap devices for apple network or with a proprietary network. However I didn't find devices for Googles find my device network so far. Thats why I came up with the idea to reprogram existing devices, to have a much bigger availibility. 

> It's worth contacting https://github.com/biemster/st17h66_FindMy

Thanks. Done. Hope we'll get an Android version soon. https://github.com/biemster/st17h66_FindMy/issues/12

pvvx, 2024-06-01:
The battery will drain quickly as frequent beacon transmissions are required.

Specifications:
https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn
```
After provisioning, the Provider is expected to advertise FMDN frames at least once every 2 seconds. If Fast Pair frames are advertised, the Provider should interleave the FMDN frames within the regular Fast Pair advertisements. For example, every two seconds, the Provider should advertise seven Fast Pair advertisements and one FMDN advertisement.
```

JsBergbau, 2024-06-02:
Thanks for taking your time going through the specification.

> The battery will drain quickly as frequent beacon transmissions are required.
Apple also transmit every 2 seconds, see https://adamcatley.com/AirTag

Since the Battery of Airtag devices and devices for Googles Find My Device last ~1year: Does that mean their circuitry is much more efficient than the Mijia Bluetooth regarding this use case?

I've found OpenHay Stack. Maybe we get Android trackable devices this way https://github.com/seemoo-lab/openhaystack/issues/245

pvvx, 2024-06-02:
> Since the Battery of Airtag devices and devices for Googles Find My Device last ~1year: Does that mean their circuitry is much more efficient than the Mijia Bluetooth regarding this use case?

Any modern BLE chip with advertising every 5 seconds with TX RF +0 DBM consumes from 6 μA or more (1..2 μA in sleep, for RF TX 6..8 mA 2 ms at +0dBm and 3..5 mA 2 ms start/run/stop CPU/SoC). 

TLSR825x (advertising every 5 seconds):
((5000ms-4ms)*0.0018mA+2ms*6mA+2ms*5mA)/5000ms = 0.00619856 mA

TLSR825x (advertising every 2 seconds):
((2000-4)*0.0018+2*6+2*5)/2000 = 0.0127964 mA

LCD driver and TH sensor in Xiaomi LyWSD03MMC constantly (in sleep mode) consume 6..10 μA.