pvvx/ATC_MiThermometer #519: MHO-C401 - display frozen

im0t0, 2024-06-01:
Hi. After [update ](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) MHO-C401 to _MHO_C401N_v47.bin_ and _ZMHOC401N_v0121.bin_ firmware display is frozen.

[image]
[image]
[image]

However, the device connected via Zigbee2MQTT without any problems.

[image]

What do I need to do to get the display to update again?

pvvx, 2024-06-01:
_[README](https://github.com/pvvx/ATC_MiThermometer): Support for Zigbee firmware is in the [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc) repository._

MHO-C401 exists in 3 versions.
As of today, Zigbee firmware only supports [MHO-C401N](https://pvvx.github.io/MHO_C401N/).

im0t0, 2024-06-01:
@pvvx можно разжевать? Т.е. у меня старая версия, но при этом встала Zigbee прошивка от новой версии, и устройство подключилось через Z2M. Но дисплей теперь работать не будет и его больше не оживить?

pvvx, 2024-06-01:
Будет работать Zigbee OTA. Можно произвести [специальную Zigbee OTA прошивку для возврата в BLE](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota).

im0t0, 2024-06-01:
@pvvx спасибо, но чтобы окончательно не окирпичить, спрошу: правильно ли понимаю, что правильная прошивка для моих датчиков это [1141-0201-99993001-MHO_C401_v46.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0201-99993001-MHO_C401_v46.zigbee) ?

И все же, у меня несколько таких датчиков. Значит я не смогу их перевести в Zigbee и чтобы при этом экран работал?

pvvx, 2024-06-01:
В Zigbee OTA файл вписаны идентификаторы. В итоге другая прошивка, от другого типа устройства, не воспримется.
Для Zigbee OTA прошивки на другое устройство требуется изменить бинарный заголовок на необходимые ID.
У вас прошит `1141-0208-01213001-ZMHOC401N.zigbee` -> подойдет только 1141-0208-xxx...

im0t0, 2024-06-01:
@pvvx спасибо, обновил по этой инструкции https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/
и далее, чтобы экран снова заработал, прошил отсюда [Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin). 
Теперь флешер корректно определяет модель, и не дает на выбор zigbee прошивку
[image]

im0t0, 2024-06-03:
Это странно, но теперь даже после обновления датчика через Mi Home на последнюю прошивку 1.0.0_0016, которая и стояла до всех манипуляций, экран тоже перестает обновляться. Т.е. датчик теперь может работать только на этой прошивке [Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin)

im0t0, 2025-01-20:
> У меня готова версия Zigbee только для указанного MHOC401N. Для других вариантов прошивок Zigbee пока нет и в ближайшие месяцы не предполагаются. Летом я больше нахожусь и работаю на природе и отдыхаю от компьютеров и прочего электронного...

@pvvx, привет. Не появилось ли прошивки для старых железок?

pvvx, 2025-01-20:
https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-0201-01233001-ZMHOC401.zigbee

closed 2025-01-20 by pvvx

im0t0, 2025-01-24:
@pvvx большое спасибо, но теперь такие датчики в Z2M отображаются вот так. Это нормально, или вопрос скорее к Z2M?

[image]

pvvx, 2025-09-03:
> Пробовал залить штатную для MHO-C410N - flasher говорит "Size firmware is more 208 kbytes!" .

Прошивка BLE для MHO-C401N v5.3 имеет размер 81.2 KB.
При скачивании бинарных файлов на github требуется выбирать "raw"  ссылку...

[image]

https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/bin/MHO_C401N_v53.bin