pvvx/ATC_MiThermometer #522: [feature request] LYWSD03MMC hide bluetooth broadcast device name ??

wazoo666, 2024-06-15:
Hi
Just a simple question.
Is it possible to hide bluetooth device name ??  Something similar than wifi AP hide setting.
I mean, once I have detected and configured on ESPHome and/or Home assistant, want to hide my bluetooth devices from mobile BTscan.
I do not want my neighbors to detect my devices.

pvvx, 2024-06-16:
To disable the display of the device in the system menu of Windows and other systems, you must disable the transmission of “AdFlags”.
Then the "Bluetooth" integration in Home Assistant will not work. The Bluetooth integration incorrectly interprets the absence of flags in the transmission. According to the BLE standard, if any data is not specified, then it should be taken by default as zero values.

"Passive BLE monitor" integration will work.

wazoo666, 2024-06-23:
thankU so much for your response.
I will check it