pvvx/ATC_MiThermometer #530: Integrate absolute humidity calculation and display it beside temperature and relative humidity

the78mole, 2024-07-31:
As relative humidity is already a good measure about moisture effect, absolute humidity (in g/m³) is sometimes much more important, since it does not change with temperature increase and decrease. In my special case, it is about the enclosure of my 3D printer and the drybox, where I would be able to directly read the absolute humidity at the device and not always need to ask my smart home GUI about.

The calculation is not very trivial, but with some simplifications, the it does not get too complex. I already wrote an article about this topic, which should be easily adaptable to the ATC firmware: https://the78mole.de/the-magic-of-absolute-humidity/

pvvx, 2024-08-02:
The CPU in the thermometer does not have floating point or floating point libraries.

the78mole, 2024-08-02:
OK, that is a valid point... Converting the calculations to fixed point is possible, but it takes some effort... If I find the time to have a deep dive into the sources and can setup the dev environment, I'll have a closer lookup.

pvvx, 2024-08-02:
No worries. I've created various dew point conversion code variations over the decades. But they all require additional battery power. For that reason, I'm not integrating them into this project.

the78mole, 2024-08-02:
Couldn't it be an option on the settings page? But also this qould require a bit of code size, even if it would not require power, if un-selected. 
Dew point would be also fine, because it is directly equivalent to the absolute humidity... and dew point is part of the BThome specification, while abs. humidity is not.

davey, 2024-08-07:
I'd also love to see either dew point or absolute humidity coming directly from device - if that's possible without too much additional power consumption (meaning reduced battery life). I wouldn't need that value too often, but once per minute would be really nice.

michapr, 2024-08-10:
@davey - if you really need it (I think in home automation), why don't you calculate it yourself?
I made it (as sample in Home Assistant) for the outdoor sensor and room sensors.

I think not every detail has to be calculated and sent by a hardware sensor. It is much more important to use energy resources sparingly in order to achieve the longest possible battery life.

the78mole, 2024-08-10:
...as I wrote in the initial message, because I would like to have the information available, when I'm standing at the measurement location. When you start printing with clodes door, the inside temperature rises and therefore, the relative humidity is also decreasing rather quickly. When you open the door, the temperature decreases even faster, so the relative humidity does. To get any reasonable information from "changing environment" in this case, you need to read it directly at that location. Same appllies to using the thermometer devices e.g., in a fridge, a drybox,... You won't always look on your phone, to have an immediate information feedback.

pvvx, 2024-08-10:
For this purpose, there is a function for displaying any values ​​on the thermometer screen. The function is [integrated into ESPHome.](https://esphome.io/components/display/pvvx_mithermometer.html)

the78mole, 2024-08-10:
That definitely sounds reasonable... Thanks!

Ircama, 2024-08-12:
I tried to add the computation of the absolute humidity to the [atc_mi_advertising GUI](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface#atc_mi_advertising-ble-advertisement-browser-gui) of the [python-interface](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface).

You can use a device supporting Python and Bluetooth to receive the advertisements from the thermometer and get the absolute humidity value.

Anyway, before pushing a contribution to the [pvvx repository](https://github.com/pvvx/ATC_MiThermometer/python-interface), I'd like to check that this addendum is correct.

@the78mole can you please verify whether [the formula](https://github.com/Ircama/ATC_MiThermometer/blob/b0fcad032a9cc79a8222a80c53aa872561424d3e/python-interface/atc_mi_interface/atc_mi_construct_adapters.py#L222-L237) is correct and that the program produces valid data?

```cmd
git clone https://github.com/Ircama/ATC_MiThermometer
git checkout ircama-absolute-humidity
cd ATC_MiThermometer/python-interface
pip3 install -r requirements.txt -r gui-requirements.txt
python3 -m atc_mi_interface.atc_mi_advertising
```

An alternative way to run it is via the precompiled *atc_mi_advertising.exe* Windows executable that can be extracted from the *atc_mi_advertising.zip* archive after downloading it from [Releases](https://github.com/Ircama/ATC_MiThermometer/releases/latest).

the78mole, 2024-08-12:
Looks good, but I'm not absolutetely sure, if 2.1674 is missing a factor of 100. In my calculation, I use 216.7.
But I can not currently test it in reality. I'm totally loaded with other work and the vacation trip is getting closer and closer.

BTW @pvvx: The "workaround" with calculation in some HA template and ESPhome-based BLE GATT displaying on the sensor does not work. ESP tells, it is sending the information to the Mi, but it does not show up on there... I also can not find any setting in the Telink Flasher Web App to enable it, only enabling/disabling battery and clock.

pvvx, 2024-08-12:
No settings are required on the thermometer.
Display is performed by [command 0x22](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f) + data structure.
[TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) has a function of demonstrating the display of arbitrary numbers and additional symbols:
[image]

Ircama, 2024-08-12:
> if 2.1674 is missing a factor of 100. In my calculation, I use 216.7.

I'll leave the code there for some time for testing/peer review. That constant looks right to me. I think yours might need fractions of humidity units, not values between 0 and 100.

> Your calculations are introducing floating point stuff and math (exp) libs. Doesn't it blow up program space and eat up battery?

No, it doesn't. This app is external to the thermometer (e.g., [Windows program](https://github.com/Ircama/ATC_MiThermometer/releases/latest)) and communicates via BLE.

---------------

An alternative way to transmit digits to the LCD is via the `-n` option of `atc_mi_config.exe` (command-line tool):

```
  -n CSV_STRING, --numbers CSV_STRING
                        Send 6 digits to LCD in the form "1,2,3,_,4,5" or "0xf5,0x05,..."
```