pvvx/ATC_MiThermometer #532: Failed when using TLSR825x USB-COM Flash Writer v0.4 to flashing TS0201_TZ3000_xr3htd96

x-magic, 2024-08-10:
I'm trying to convert my [Zigbee TS0201_TZ3000_xr3htd96](https://pvvx.github.io/TS0201_TZ3000/) into BLE with CFW. 

First I tried to do Zigbee OTA with info from #529 however can't make Z2M to OTA the firmware. So instead I decided to flash the device with UART instead. Here's my wiring: 
[image]
(wired based on [this](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp) and [this](https://github.com/pvvx/BLE_THSensor#%D0%BF%D0%B5%D1%80%D0%B2%D0%BE%D0%BD%D0%B0%D1%87%D0%B0%D0%BB%D1%8C%D0%BD%D0%B0%D1%8F-%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0-%D1%83%D1%81%D1%82%D1%80%D0%BE%D0%B9%D1%81%D1%82%D0%B2%D0%B0))

Then I used the [USB-COM flasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) (460800, 3sec, with a CH343 UART), erased all flash, then flashed `TS0201_v48.bin` from [pvvx/ATC_MiThermometer](https://github.com/pvvx/ATC_MiThermometer/blob/master/TS0201_v48.bin) but no luck (after soft reset MCU, the LED no longer lights up even when button is pushed or replacing battery). Can't find the device in either Zigbee nor BLE. 

Also tried to flash `TS0201_v47.bin` from [pvvx/BLE_THSensor](https://github.com/pvvx/BLE_THSensor/tree/master/source/TS0201/bin), same thing. 

Finally tried to erase and flash the FullFlashbin from [pvvx.github.io/TS0201_TZ3000](https://pvvx.github.io/TS0201_TZ3000/), the device won't restore back to Zigbee mode. Still no LED on battery replacement nor button push. 

Basically at this stage it's bricked. 

Am I missing any steps? Any way I can be sure the firmware is properly flashed? Any suggestion is appreciated! 

(Doesn't really matter for this one thermometer - I'm experimenting on it. Just that I have two more that I actually need them to be on BLE mode as Tuya Zigbee firmware's refresh rate is super slow, like 30 minutes per refresh mostly, while my LYWSD03MMC are refreshing at 30s or quicker)

Thank you!

pvvx, 2024-08-17:
Connect UART RTS to the RST pin of the ZTU module.

x-magic, 2024-08-18:
> Connect UART RTS to the RST pin of the ZTU module

Yes, that's how I connected mine. RST on ZTO to RTS on UART. 

Is it normal that pin B7/B1 (UART RX/TX) produces no output when booting? My issue is that I can't tell if the flashing is successful or not. All I know is that before I flash the firmware through SWS, `Soft Reset MCU` works and it toggles the device and the LED indicator flashes when booting or when button is pushed. Now I can't tell if the device is booting or not, as the LED or button are not responsive. 

Thank you

pvvx, 2024-08-18:
There is a simple version of the programmer with reading:
https://github.com/pvvx/TlsrComSwireWriter - [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)

x-magic, 2024-11-29:
> There is a simple version of the programmer with reading: https://github.com/pvvx/TlsrComSwireWriter - [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)

Thank you. I have wired up everything according to the schematic, with 1.5k resistor on TX line. 

It seems both reading and writing works as follow: 

```
C:\TlsrComSwireWriter>python TLSR825xComFlasher.py -p COM7 -b 921600 -t 70 wf 0 TS0201_v47.bin
=======================================================
TLSR825x Flasher version 00.00.05
-------------------------------------------------------
Open COM7, 921600 baud...
Reset module (RTS low)...
Activate (70 ms)...
UART-SWS 92160 baud. SW-CLK ~24.4 MHz(?)
Inputfile: TS0201_v47.bin
Write Flash data 0x00000000 to 0x00012a04...
-------------------------------------------------------
Worked Time: 35.023 sec
Done!

C:\TlsrComSwireWriter>python TLSR825xComFlasher.py -p COM7 -b 921600 -t 3000 rf 0 80000 read4.bin
=======================================================
TLSR825x Flasher version 00.00.05
-------------------------------------------------------
Open COM7, 921600 baud...
Reset module (RTS low)...
Activate (3000 ms)...
UART-SWS 92160 baud. SW-CLK ~24.4 MHz(?)
Outfile: read4.bin
Read Flash from 0x000000 to 0x013880...
-------------------------------------------------------
Worked Time: 82.873 sec
Done!
```

However after flashing the binary, contents read back from the flash is in such a pattern that first 0xff of every 0xfff block is corrupted: 
[image]

Doesn't matter if I change baud or time of activation. Same results. 

I also tried to erase flash first then read the flash, and correctly read an empty flash with only FF. Then I tried to write 00 to the whole flash (1MB) and read back, still getting FF for the first 0xff of every 0xfff block: 
[image]

Any suggestions? Thank you!

pvvx, 2024-11-29:
Problematic Flash?
Doesn't correctly return Write In Progress (WIP) bit to Status Register?

[TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)
For testing, I added a delay after Erase Sector. And command `we` - Write file to Flash without sectors erases.

x-magic, 2024-11-29:
> Problematic Flash? Doesn't correctly return Write In Progress (WIP) bit to Status Register?

Fortunately it's not a dead flash! Still not sure if WIP bit is the reason - not really my strong suit. 

> Is +3.3V (and GND) power supply connected?

No, I was using the battery to power the module. Thank you for your advice - I've connected the VCC and let my serial adapter to power the module instead. 

> [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py) For testing, I added a delay after Erase Sector. And command `we` - Write file to Flash without sectors erases.

Thank you for updating the flasher! I used 're' command to erase the flash first, then flashed 'ts0201_orig_ff.bin' and the sensor is working again in Zigbee mode! 

I couldn't make Zigbee OTA work with my setup, so I eventually downloaded Tuya2BLE firmware from https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota, removed first 0x3d bytes (seems to be a header of Zigbee OTA?) and flashed to the module, and it worked! 
[image]

Now I need to check if the report frequency is indeed higher on BLE...

Thank you very much for all the support!

pvvx, 2024-11-30:
> removed first 0x3d bytes (seems to be a header of Zigbee OTA?) and flashed to the module, and it worked!

https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/TS0201_v48.bin

x-magic, 2024-11-30:
Must have missed it. Thank you 🙏

closed 2024-11-30 by x-magic