pvvx/ATC_MiThermometer #533: Count/switch not visible in advertisments

vahempio, 2024-08-13:
Hi,

(First, thank you for this incredible work!)

On a LYWSD03MMC, I tried to have the pulse counter from a reed switch connected to P8. According to the Readme.md (https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#reed-switch-or-button-functions), there is nothing to do for having the value in advertisments (I don't need "events" on pulses).

But there is no traces of pulse counter in advertisments, according to nrfconnect (i decoded only temp/hum/voltage/power, no count value). Even with the activation of events, there is no changes in the adv.

[image]

Question: how to enable counter value in adv? Thanks a lot!

pvvx, 2024-08-13:
Support for receiving and transmitting a reed switch counter is only available in the [BTHome](https://bthome.io/) standard.

In [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) you need to set the reed switch usage option:
[image]

vahempio, 2024-08-13:
Thank you for the fast answer!
My config was correct, but I was on the 4.7 fw version: after an upgrade to 4.8 beta fw, it works like a charm, thank you very much!

closed 2024-08-13 by vahempio