pvvx/ATC_MiThermometer #534: LYWSD03MMC-z lost settings after update

i4mr000t, 2024-08-17:
Hello,

today I received an update notification from Homeassistant for all my LYWSD03MMC-z devices. I tried this update with one device after the update was successfully the devices resets the calibration settings and the icon on the display which was previously disabled enabled. 
I tried to disable the icon on the display first and recognised that I cannot change any value from settings menu in z2m.

when I try to change the icon on the display I receive the following error:

Error 2024-08-17 17:22:27z2m: Publish 'set' 'comfort_display' to 'Temperatursensor Ankleidezimmer' failed: 'Error: ZCL command 0xa4c1381b6dd38292/1 hvacUserInterfaceCfg.write({"2":{"value":0}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot write USE_DATA_TYPE without dataType option specified)'

Devices without the new Update:
[image]

device with the new update:
[image]

gurrier, 2024-08-18:
[image]
I see the update offered but have not installed it yet.  Following here.

i4mr000t, 2024-11-09:
Any Informations on this?

The Device ist still not working. On every HA Restart the Update Show up but i don’t install it on other devices. 

Btw all my other devices which worked correct in the past are now not working anymore except sending entity states. 
I have not touched anything.

i4mr000t, 2024-11-24:
I tried a new Sensor. 

Flashed Firmware successfully as in the repo showed.

Device is recognised in z2m and receiving data but its not possible to set the smiley to hide. I receive following error:

z2m: Publish 'set' 'comfort_display' to '0xa4c138fbe5fcc3dc' failed: 'Error: ZCL command 0xa4c138fbe5fcc3dc/1 hvacUserInterfaceCfg.write({"2":{"value":0}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (ZCL command 0xa4c138fbe5fcc3dc/1 hvacUserInterfaceCfg.write({"2":{"value":0}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (ZCL command 0xa4c138fbe5fcc3dc/1 hvacUserInterfaceCfg.write({"2":{"value":0}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Cannot write USE_DATA_TYPE without dataType option specified)))'

i4mr000t, 2024-12-02:
I think I solved it.

Problem was that I was using a converter. 

I deleted my converter and can hide or show the smiley. After that i tried to update from 3001-0120 -> 3001-0123 the device was not working after that but i unplugged and replugged the batterie. Now the device is working correctly.

Actually I´m trying to Upgrade all other devices.

3001-0120 -> 3001-0123 (no converter)
3001-0121 -> 3001-0123 (no converter)

[image]

i4mr000t, 2024-12-03:
Problem Solved.

I Installed  3001-0123 on all Devices without a Converter and everything is working as expected.

closed 2024-12-03 by i4mr000t