pvvx/ATC_MiThermometer #536: CGD1

redblue-pkt, 2024-08-29:
I’m currently working on a project that requires extracting or dumping OTA (Over-the-Air) firmware from a device CGD1. I've reviewed some documentation and attempted various methods, but I’m still facing challenges with the process. Could someone please provide detailed instructions or recommend tools and techniques for dumping OTA firmware?

pvvx, 2024-08-30:
It is possible to read the firmware using a programmer.
OTA function does not support reading program dump.

With OTA you can download a new program that will not erase the previous code.
Using this new program, read the old OTA program. Then restore the pointer to the start of the old OTA version.
https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker

redblue-pkt, 2026-01-16:
ALARM CLOCK
TLSR8253 F512ET32 ZTL2115
XT25F08B-S 2050THOD
BU9792AFUV
ANALOGTEK AT8372AT 2028B

redblue-pkt, 2026-10-27:
Hi @pvvx,

I've been working on an experimental CGD1 alarm clock port based on your ATC_MiThermometer firmware.

I'm making my current work publicly available, even though it is still a work in progress and probably contains quite a few bugs, implementation mistakes, and things I've overlooked.

In my own testing, most of the functionality appears to work, but that's only my personal assessment. The audio codecs in particular need more testing, and the entire implementation would benefit from a proper technical review.

I would really appreciate your expert opinion on the code and overall implementation. There may be problems or better approaches that I'm simply not seeing.

I'm also considering extending the clock beyond its original functionality, particularly in the audio area. Possible future directions include a simple audio player and playing sounds or notifications from Home Assistant, although these are only ideas at this stage.

I'm including the modified sources, firmware, web interface changes, and the original Qingping firmware (1.0.1_0130) for reference. The original firmware also provides a potential route back to stock via OTA, but that recovery path still needs to be verified.

I'd be very happy if you or other experienced developers could review the work, improve it, or take over further development.

Thanks again for your original project and the work you've put into it!

[model-v1_cgllc.clock.dove_original_1.0.1_0130_dfbb607c08.bin.gz](https://github.com/user-attachments/files/33996385/model-v1_cgllc.clock.dove_original_1.0.1_0130_dfbb607c08.bin.gz)

[CGD1_firmware_web.patch.gz](https://github.com/user-attachments/files/33996388/CGD1_firmware_web.patch.gz)

[CGD1_sources.patch.gz](https://github.com/user-attachments/files/33996390/CGD1_sources.patch.gz)