pvvx/ATC_MiThermometer #538: [Suggestion] Dynamic Sensor Data Notifications

k-korn, 2024-09-08:
Currently the firmware just sends temperature and humidity at pre-configured interval.

In my similar projects with ESP32-based sensor I've used a dynamic approach which reduces the amount of data points sent to HA while maintaining precision. It can be described as follows:

- Let's define a sending threshold (say, 0.1C for T and 0.5% for H) and max send interval (say, 300 seconds).
- Values are measured (but not sent) at regular intervals, as previous.
- If the value is within +- threshold from the last value sent, and if max send interval (from the last time value has been sent) did not pass - do not send the value via BLE.
- If the value changed for more than the threshold, send the value and update time it was sent and update the last sent value.
- If the value did not change significantly, but max send interval passed - also send the value and update time it was sent and the last sent value.

This way, if the value changes fast we will still get frequent upgrades and fast sensor response. And if it changes slowly - we will get 1 value each 300 seconds, but we do know that it changed no more than the threshold from the last one.

This approach worked fine for me for a couple of years and will be great to see it implemented here.

I think I can actually produce the pull request for it, but would need a hint on where in the code this logic is implemented.

pvvx, 2024-09-08:
1. How to connect to a BLE device that does not send BLE advertise? 
2. Or when the advertise period is more than 10 sec (maximum by standard = 10 sec)?

k-korn, 2024-09-09:
OK, I was not aware BLE ***needs*** to advertise every X seconds or the recipient does not get anything - and this might be a deal breaker here.

Anyway, I've used this approach for many years with MQTT-connected devices and it works just fine.
> Such an interval makes no sense in automation (IoT, Smart Home).

We are not sacrificing response time or accuracy when setting "idle" interval to 300, or even 600 seconds.
Most automations trigged on "when temperature crossed X threshold", and in our case if the value is constantly rising you will get the fastest response possible, as the temperature will most likely rise fast past the threshold. 

Here is an example:
[image]
At around 07:20, when the value rapidly changes, measurements are emitted as fast as possible. And then around 8:20 - 8:40, when the changes are minimal, the value gets updated every 10 minutes - and this is still fine.

As for the threshold values - these can be set lower or higher, of course. It was just my gut feeling that temperature  changes below 0.1C are not significant for human perception anyway.

pvvx, 2024-09-09:
> It was just my gut feeling that temperature changes below 0.1C are not significant for human perception anyway.

For some automations, it is not the absolute value that is important, but the trend of change (derivative).