pvvx/ATC_MiThermometer #542: Offset and Slope factor issue

octane321, 2024-09-17:
hi there, and first of all many thanks to your work - i love it!

But i´m struggling getting 10 sensors equalized in temp and humidity reading.
Trying to use only offset, only slope or both combined gives me worst equalized sensors than before..
So i dived a bit into the calculations and found out, that "calibration" with offset and slope is really annoying and need a high effort if you want to do it the right way.

My personal opinion for temp. "normalization" and i´m still on research (same again for humidity):

Step A: Correct deviation on linear behavior:
1st: One will need all sensors measurements at a defined low temp (let say 5°C) e.g. in the fridge
2nd: One will need all sensor measurements at a defined high temp (let say 40°C) e.g. in the oven
3rd: Pick one sensor as reference (same sensor for temp & humidity!)
4th: calculate the range of all sensors
5th: calculate the deviation of all n sensors range-difference in relation to reference sensor
6th: calculate the new slope factor and set it in all sensors
Result: All sensors should have quiet the same changes when temp and hum changes - but the measured temp. are still all different

Step B: Correct the offset
1st: One will need again need all sensors measurements at a defined low OR high temp like above
2nd: calculate the temp deviation of all n sensors range-difference in relation to the reference sensor
3rd: add the deviation to the default offset (-45°) and set up all sensors

It would be much more easier to take only a high and low measurement per sensor and put that into a calibrate linear function with a setpoint value for low and high value
Do you see any chance getting this implemented into the firmware?

Thanks in advance and have a great time!

octane321, 2024-09-17:
should be calculated like this:

Tcali =
((T_high_meas - T_low) / (T_high - T_low)) * x  +  (T_high_meas - (((T_high_meas - T_low_meas) / (T_high - T_high_meas)) * T_high))

pvvx, 2024-12-19:
The source codes are given. Make corrections and build your own firmware with the formulas you need.

galileo-pkm, 2025-04-13:
@Schneelocke Thank you for the formula explanation. This should be documented somewhere. Project Wiki?

Here is a short script to calculate the offset and slope, run it at programiz.pro/ide/python/ if you can't be bothered to install python dependencies. 

```
from sympy import symbols, Eq, solve

def calibrate_sensor():
    print("Enter calibration points (minimum 2 required)")
    
    # Get number of calibration points
    while True:
        try:
            num_points = int(input("\nNumber of points: "))
            if num_points >= 2:
                break
            print("Minimum of 2 points is required")
        except ValueError:
            print("Please enter a valid number.")

    # Get current calibration parameters
    current_offset = float(input("\nCurrent offset (e.g., 0.0): ") or "0.0")
    current_slope = float(input("Current slope (e.g., 100.0): ") or "100.0")

    # Collect all calibration points
    points = []
    for i in range(num_points):
        print(f"\nCalibration Point {i+1}:")
        displayed = float(input("Displayed sensor value: "))
        actual = float(input("Actual reference value: "))
        points.append((displayed, actual))

    # Calculate normalized raw values
    raw_values = [(d - current_offset) / current_slope for d, _ in points]

    # Symbolic equation solving
    O, S = symbols('O S')
    equations = []
    for (_, actual), raw in zip(points, raw_values):
        equations.append(Eq(O + S * raw, actual))

    # Solve the system (using least squares if > 2)
    if len(equations) == 2:
        solution = solve(equations, (O, S), dict=True)[0]
    else:
        from sympy import Matrix
        A = Matrix([[1, r] for r in raw_values])
        b = Matrix([[a] for _, a in points])
        solution = (A.T * A).inv() * A.T * b
        solution = {O: solution[0], S: solution[1]}
    
    print("\n=== Calibration Results ===")
    print(f"New Offset: {float(solution[O]):.6f}")
    print(f"New Slope: {float(solution[S]):.6f}")
    
    print("\n=== Verification ===")
    for i, ((displayed, actual), raw) in enumerate(zip(points, raw_values)):
        calibrated = solution[O] + solution[S] * raw
        print(f"Point {i+1}: Displayed {displayed:.2f} → Calibrated {float(calibrated):.2f} (Actual: {actual:.2f})")

if __name__ == "__main__":
    calibrate_sensor()
```

pvvx, 2025-04-13:
The coefficients are taken from the sensor datasheet.
For example for SHTv4
https://sensirion.com/media/documents/33FD6951/67EB9032/HT_DS_Datasheet_SHT4x_5.pdf

[image]

But the TLSR825x chip does not have an FPU and a fast division command. Therefore, the slope coefficient for calculation is taken in a fixed-point format [16.16] and the work is done by multiplication, followed by discarding the lower 16 bits.
And the result should be in 0.01%.
Therefore, the firmware uses Z: --600 and S: 12500.

galileo-pkm, 2025-04-13:
The device that I have (manufactured 2024.11) has the SHT3 sensor. At 75%RH it showed 71 and at 95%RH 90. Temperature was much closer to real value, +0.7C difference.

pvvx, 2025-04-13:
> The device that I have (manufactured 2024.11) has the SHT3 sensor. 

SHTv3 = [SHTC3](https://sensirion.com/products/catalog/SHTC3).
https://sensirion.com/media/documents/643F9C8E/63A5A436/Datasheet_SHTC3.pdf

[image]

Temp Z: -4500 S: 17500 
Humi Z: 0 S: 10000

galileo-pkm, 2025-04-14:
I presumed that the formula is the same for both sensors ...