pvvx/ATC_MiThermometer #544: Support for Meawow MHO-C402 (frozen display)

Gucioo, 2024-09-17:
Hi,

I bought Meawow MHO-C402. With stock firmware it was not possible to connect with BT.
After checking PCB I found out electrical connections seems to be the same as MHO-C401.
Only 3 vias on the PCB were routed a bit different, moved 1-2 mm.

You did great work on the firmware and using your instructions I was able to download modified firmware:
 **MHO_C401_v48.bin to Meawow MHO-C402** using [USBCOMFlashTx](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) connecting Vcc, Gnd, Tx, Rst.

**Now it is possible to connect with BT, but the display is frozen**. Do you have any idea, what could be wrong or which firmware could work with it? How to enable eink display?

[image]
[image]
[image]
[image]

pvvx, 2024-09-18:
The display controller and the display itself can be different. This is the main difference between different models.

Gucioo, 2024-09-19:
It is like **MHO-C401**. After going thru the topic I downloaded few other firmware's and it **started to work again on Original_OTA_Xiaomi_MHO_C401_v1.0.0_0016.bin**.     
Tried re flashing custom again, but display was still frozen on MHO_C401_v48.
It doesn't blink after custom download. After original is downloaded screen blinks and displays temp.

If I will find time I may get flash / display mapping as you mentioned in the other topic.