pvvx/ATC_MiThermometer #545: Displaying arbitrary data on the LCD

davystrong, 2024-09-19:
Hello! I would like to use one of these displays to show the temperature at a different location (using a separate Zigbee sensor). Can I transmit arbitrary data to these devices over Zigbee? I believe [this section](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#interface-for-receiving-and-displaying-data-on-the-lcd) indicates that I can, but I'm not sure. If so, would all devices be compatible?

pvvx, 2024-09-19:
There is no such function for the Zigbee version.

pvvx, 2024-09-20:
The example described in https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#interface-for-receiving-and-displaying-data-on-the-lcd is in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

ESPHome also has HA integration for this feature.

davystrong, 2024-09-20:
Great! Thanks so much for your help! I guess I can close this issue now.

closed 2024-09-20 by davystrong