pvvx/ATC_MiThermometer #546: Update from 3.2 to 4.7 failing

devzwf, 2024-09-21:
Tryin to update from 3.2 to 4.7 
and i am receiving 
Status: Update error: NotSupportedError: GATT Error Unknown.

10:13:23 PM: Hardware Revision String: B1.4
10:13:23 PM: Software Revision String: V3.2
10:13:23 PM: Firmware Revision String: github.com/pvvx
10:13:23 PM: Detected custom Firmware
10:13:23 PM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.2
10:13:23 PM: Custom config HEX string: 55250000002804a97c31383c000000
10:14:09 PM: Start DFU
10:14:40 PM: Update error: NotSupportedError: GATT Error Unknown.
10:14:40 PM: Disconnected.

pvvx, 2024-09-21:
1. Replace the battery
2. Set "Connect latency" 500..1000 ms
[image]

manelrodero, 2024-12-06:
Hi @pvvx,

I'm trying to upgrade from version 4.3 to latest 4.8 and I get the same error: _Update error: NotSupportedError: GATT Error Unknown_.

I've restarted my phone several times.
I've restarted Chrome and I've using a single tab with your TeLink Flasher web page.
I've replaced the battery with a new one at 99% (previous was at 53%).
Latency is the same as you mentioned (1000 ms)

```
22:19:23: Reconnect
22:19:28: Hardware Revision String: B1.6
22:19:28: Software Revision String: V4.3
22:19:28: Firmware Revision String: [github.com/pvvx](http://github.com/pvvx)
22:19:28: Detected custom Firmware
22:19:31: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.3
22:19:31: Custom config HEX string: 55a6100000500ca931ff04b4
22:19:44: Load firmware file 'ATC_v48.bin'...
22:19:44: File: ATC_v48.bin
22:19:44: File size: 83092 bytes
22:19:44: Count: 5194
22:19:49: Start DFU
22:19:55: Update error: NotSupportedError: GATT Error Unknown.
22:19:55: Disconnected.
```

What are next steps to flash? I remember it was so easy long time ago when I flashed from stock to your 4.3 :-/

Thanks.

pvvx, 2024-12-06:
Everything seems to be working fine:
```
00:30:39: Searching for devices
00:30:59: Connecting to: LYWSD03MMC
00:31:05: Hardware Revision String: B1.4
00:31:05: Software Revision String: V4.8
00:31:05: Firmware Revision String: github.com/pvvx
00:31:05: Detected custom Firmware
00:31:05: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.8
00:31:05: Custom config HEX string: 55831000002804a9313100b4
00:31:17: Load firmware file 'ATC_v48.bin'...
00:31:18: File: ATC_v48.bin
00:31:18: File size: 83092 bytes
00:31:18: Count: 5194
00:31:25: Start DFU
00:31:52: Update done after 26.883 seconds
00:31:56: Disconnected.
```
Try connecting from another device.

manelrodero, 2024-12-06:
I've tested another unit with the same 4.3 firmware and I've updated to 4.8 successfully ;-)

So, what's the problem with the first one? The only different thing is the config (because I see a different HEX string). So I'll compare both configs to see the diferences.

Thanks.

```
22:30:23: Searching for devices
22:30:34: Get Advertising MAC (Web Experimental Features!)
22:30:34: Connecting to: ATC_<Redacted>
22:30:36: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 103, CtrID: 0x5850 Not bound, Standard certification, MAC: <Redacted>, Temp: 19.6°C Humi: 65.6%
22:30:42: Hardware Revision String: B1.6
22:30:42: Software Revision String: V4.3
22:30:42: Firmware Revision String: [github.com/pvvx](http://github.com/pvvx)
22:30:42: Detected custom Firmware
22:30:44: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.3
22:30:44: Custom config HEX string: 55a61000005004a9316404b4
22:30:48: Start DFU
22:31:49: Update done after 60.825 seconds
22:31:53: Disconnected.
```

pvvx, 2024-12-06:
The configuration of different versions of fw is different.

manelrodero, 2024-12-06:
Thanks for replying so quickly.
I've checked the settings on both units.

I've set the unit that has firmware 4.3 and fails to update to the same settings as the unit I was able to upgrade to 4.8.
Basically they differed in the Measure Interval and Minimum LCD refresh.

And nothing. It still won't update.

I've also tried a different phone and a different browser (Edge instead of Chrome) with the same result.

The only difference I see on your website is that when I connect to the unit that was able to update, the "Sensor Settings" section appears (where the Slope factors are). However, this section does not appear on the unit that won't update.

Could this be a clue?

manelrodero, 2024-12-06:
Another difference I've seen.

In the "Configuration" section, after pressing the "Get Config" button, on the unit that won't update there are some fields that aren't on the unit that has been updated.

These fields are:
- Temperature offset: 0.0º
- Humidity offset: 0.0%

They are just before the BT5+ PHY fields, etc.

pvvx, 2025-01-02:
> In the "Configuration" section, after pressing the "Get Config" button, on the unit that won't update there are some fields that aren't on the unit that has been updated.

The coefficients for the sensor have moved to another setting. This is the difference between the old and new versions. 
New versions use extended coefficients...