pvvx/ATC_MiThermometer #547: LYWSD03MMC not connectable

phako, 2024-09-26:
I have flashed a LYWSD03MMC with v47 BTLE firmware. Shortly after flashing I can connect to the sensor for configuration, but after a short while I can no longer connect to it, nor getting any measurements out of it. 

The display updates fine. The first update was done by OTA from vendor fw to yours, in an attempt to recover the device I did a serial flash through the UART web flasher.

Is the sensor just broken? Is there any way to get some debugging out of the device?

bzgec, 2024-09-26:
I had similar problem when I set advertising interval too big (to 10 seconds). I could not connect to LYWSD03MMC from Linux PC (authentication timeout), but I could connect to sensor from Android phone. I just changed it back to default value and I could connect to sensor from Linux PC.

I have just found a note in README which could be another solution:
> On Linux, when installing large advertising intervals, change the file '/etc/bluetooth/main.conf', variables: `LEScanIntervalAutoConnect=` `LEScanWindowAutoConnect=`. Value in ms.

phako, 2024-09-27:
I am using an ESPhome btproxy, not Linux. Settings are default, I did not change anything. And also the device is not visible at all, that apparently is cached in the btle scanner I used previously.

jksmurf, 2024-10-17:
I have the exact same issue with one of my two devices, both HW v 1.4 LYWSD03MMC with ATC_v47.bin BTLE firmware.

- One of the devices, which I flashed OTA from a **non** 2.1.1_0159 FW is working fine (and integrates into HA using BTHome).

- However the device that just doesn't connect to BTLE (but shows up just fine on a scanner) originally had 2.1.1_0159 FW, which I downgraded to the Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin), then flashed to ATC_v47.bin using pvvx [Telink Flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html). 
- This went fine, although often to "see" the current config (and amend it) I have to click connect in the flasher, it gets stuck, so I press reconnect and it comes up with the normal screen wtth the blue boxes for flashing various FW.

I use a Xiaomi Gateway 2 flashed to BLE using 3rd Part FW as my Bluetooth Gatweway but I also tried adding a BT Adapter direct to my Rpi4 running HA, without an improvement.

I have flashed and reflashed various FW but would love to have some ideas how to troubleshoot or what to test next? 
I do not need to go to the Zigbee, prefer the BLE one.

[image]

phako, 2024-10-27:
I experiemented a bit (basically trying to go to zigbee which also did not work), The device is connectable and configurable in the flasher if I connect 3.3V (from UART or similar). With the battery in, no dice. the battery has 3.0something volts.

phako, 2024-10-27:
Hm.  I think this is actually a problem with the bluetooth proxy and it being completely overwhelmed by the number of devices available in my environment

jksmurf, 2024-10-27:
Maybe same issue as me ? 
Simple solution?

https://github.com/pvvx/ATC_MiThermometer/issues/559

phako, 2024-10-27:
No, it's different. BTHome is working in principle, I have more sensors that just work, one that is stubborn

jksmurf, 2024-10-27:
Ok. I presume you have reflashed up and down to original and a few times to check ?

phako, 2024-10-27:
yes. It is now available all the time in random btle scanners and the OTA flasher, just not seen on the esphome btproxy. So this can be closed

closed 2024-10-27 by phako