pvvx/ATC_MiThermometer #548: LYWSD03MMC not connecting to Home Assistant after Battery was replaced after depletion

makerwolf1, 2024-09-27:
I've been using an LYWSD03MMC (flasher readout: LYWSD03MMC B1.4, Software Version: 4.7, Sensor: SHTC3 (SHTV3)) for one full battery duration with the original ATC firmware and home assistant with absolutely 0 problems. Now the battery depleted completely and after replacing the battery, only the RSSI value would update. I figured I might be able to solve it by just deleting the device and adding it again, unfortunately home assistant does not recognize it. 
Then I tried updating it via the firmware from the original ATC version, same issue. I then uploaded the original firmware and the device was discovered by HA immediately. I then tried flashing the new firmware form this repository and everything works fine, except it is not discovered in Home Assistant. 

To make matters even a little bit more interesting, I still had one more unit sealed in the box, went through the same procedure and it was discovered just fine in home assistant with firmware 4.7. 

Any idea what the issue could be and how to troubleshoot?

pvvx, 2024-09-27:
Most likely, something related to the sensor fell off the board.
Disassemble, carefully wash the board, blow/heat the board with a soldering hair dryer...

makerwolf1, 2024-09-28:
I would agree with you if the display would not display temperatures fine, also your webflasher shows temperatures just fine. Did not include that info, sorry. 

Really the only problem seems to be home assistant not detecting the device.