pvvx/ATC_MiThermometer #554: Firmware compatibility with CGG1-T ?

vincedarley, 2024-10-07:
Hi - I have a few of these "T" versions of the QingPing sensor, which can run over bluetooth or Thread, and support HomeKit. Are you aware if the custom firmware works on them? (And whether it could support Thread)?  Thx

pvvx, 2024-10-08:
I don't have CGG1-T and I don't know what chipset is used.

vincedarley, 2024-10-08:
If it helps, there are teardown pictures of it here:
https://device.report/m/0857ffc50bc59be35306e9dac1fd1874658586fce5e9a0592d30c2f91a73bd46.pdf

Particularly the last picture is close enough to see writing on the main
chip. Vince

[image: image.png]

vincedarley, 2024-10-08:
Looks to be N52840, which is a version of this
https://www.nordicsemi.com/Products/nRF52840 I think.

Vince.

pvvx, 2024-10-08:
nRF52x chips have OTA firmware locked with a key.
Use JTAG.

closed 2024-12-19 by pvvx