pvvx/ATC_MiThermometer #557: noise on Temperature of LYWSD03MMC

axlerose, 2024-10-14:
I've 3 LYWSD03MMC that report a lot of noise on temperature value, it's a sensor problem or firmware/settings problem?

[image]

pvvx, 2024-10-15:
This is a settings problem.
[image]
For some types of sensors it is advisable to disable this option. 
You did not write what version of LYWSD03MMC and what firmware is used...

jj-uk, 2024-11-08:
3 out of 7  devices have started reporting noisy readings after working perfectly for around a year or so. 
Its suspicious that they were all flashed at the same time, and all started having issues at the approx same time. 

I'll take a look at the settings mentioned above.