pvvx/ATC_MiThermometer #559: Lesson learned regarding Xiaomi LYWSD03MMC not being added as BTHome Device in HA

jksmurf, 2024-10-18:
Hopefully this will help someone and wil be left here, I appreciate it is not 100% ATC FW related but if you did what I did and tried literally hundreds of FW flashes, downgarding to original Xiaomi Mi FW, updating to ATC_38 through ATC_48, trying all for advertising modes, trying Zigbee then back to BLE, and you STILL couldn't add a device to Home Assitant via BTHome, maybe you did what I did.

I have two identical HW 1.4 LYWSD03MMC H&T devices, one added to HA BTHome just fine, the other rmust have been added at soem point, but I deleted it and it was never to be seen NOR ABLE TO BE ADDED to HA, ever again. 

I knew it worked as I could see it advertising itself when I checked bluetoothctl devices in a HA terminal and it added fine using Miijia advertsiing mode (Xiaomi BLE). As above, I though it was a bad flash, wroing advertising mode, integration wrong, settings wrong, a bad device. It was none of these. 

So after four days, I found a FIX ... the issue is NOT that the device flashed with e.g. custom firmware LYWSD03MMC from pvvx (or whoever) using BTHome as the advertising type is not working in your Xiaomi H&T Sensor (all text for future searchers...), it is simply that at some stage you (like me) have "Dismissed" or *ignored* adding a device and BTHome simply remembers that. 

Note that this applies to other integrations where things do not get added back in or found again... 

So:

1. Fire up [WinSCP ](https://winscp.net/eng/download.php), (assuming you have FTP Access to your HA IP Address). I attach some screenshots from the Terminal & SSH Addon but will leave you to figure that out.
2. Navigate to /homeassistant/.storage and find the file core.config_entries.
3. Open it in a text edtor and carefully (without deleting or changing any other text) search for either *BTHome* (or if you are having other items ignored, try *"source": "ignore"* (this one with quotes "") or just *ignore*.
4. Amend *ignore* (provided it is a bluetooth device that is being ignoed !!) to *bluetooth*.
5. SAVE the file. Exit WinSCP. Exit HA or restart it. Go back in and check the BTHome integration. it shoudl have added your device back in.
6. IF in the interim you added the device using a DIFFERENT advertising type (or integration), it MAY show the device under BTHome as having TWO integrations AND ALSO under e.g. Xiaomi BLE, TWO integrations. Just delete the oine you do not want (in my case Xiaomi BLE) and oit will shows a single advertsining type device under BTHOME.

HTH as it has taken me four days to find this out; I thought my device was stuffed. It wasn't. HA was just doing what you told it to do... and forgot :-).

k.

[image]
[image]
[image]
[image]
[image]

pvvx, 2024-10-18:
[image]

jksmurf, 2024-10-18:
> **1. Settings -> Devices & services:** ![

lol, that’s certainly one way of doing it, thank you. Oddly though I don’t think I  saw that option on my system. 

In any case it’s a good top 🙏

g6094199, 2024-11-05:
anyother easy way is: take a esp32 and flash ESPhome with bluetooth proxy function. add the esphome device to HA and bthome units get visible automaticly.

pvvx, 2024-12-19:
> anyother easy way is: take a esp32 and flash ESPhome with bluetooth proxy function. add the esphome device to HA and bthome units get visible automaticly.

ESP32 has Bluetooth version 4.2 and does not accept all BTHome variants that are available in alternative firmware - Bluetooth version above 5.0 is required. And also does not work with a large number of BLE devices. That is, ESP32 is not recommended for use with BLE devices.

g6094199, 2024-12-19:
this may be but for me its the method that simply works out of the box. everything else i  tried was a nightmare and not working in the end.
im my case bthome is working rock stable on two different proxies in 2 locations connected via vpn to HA. dont remember exactly but 1 proxy should be a esp32 and one a esp32-S. both working stable, even if there where some dropouts in BL reception it accurate enough for most szenarios. i dont recognize any problems so far.

pvvx, 2024-12-19:
> in most cases it doesnt need to be perfect. i prefer simplicity over perfection. live is short.

It couldn't be simpler - solving the problem requires two mouse clicks in the [Home Assistant menu](https://github.com/pvvx/ATC_MiThermometer/issues/559#issuecomment-2422958766).
And not buying outdated chips, power supplies for them, pulling wires, programming and connecting all sorts of nonsense...