pvvx/ATC_MiThermometer #560: TRG pin constantly firing after first press

davystrong, 2024-10-18:
I have an MHO C401 (possibly knock-off). I'm running into an issue when trying to configure TRG to report immediately in connected mode. I set the RS mode to Switch in the Telink Flasher, then disconnect and connect using a Python program. In this program I write 0x33ff to 0x1f1f then listen for notifications on 0x1f1f. This correctly notifies me every 20 seconds giving me the temperature, humidity, etc.

However, when I click the button, it instantly starts firing 0x4504 and 0x4505 events non stop (depending whether the button is currently help down). Am I doing something wrong? Do I need to clear something? I imagine that continuously firing this even will ruin the battery life.

My RS report interval is 3600 sec and my advertise interval is 2500.0 ms. These may be relevant though I don't really understand what either of these mean.

pvvx, 2024-10-19:
This has been fixed in version 4.8.

davystrong, 2024-10-19:
That's excellent, thank you! Tested and it works correctly now.

closed 2024-10-19 by davystrong