pvvx/ATC_MiThermometer #562: MY18B20 sensor

user784213, 2024-10-25:
Hello. In the source code of the firmware I found a mention of the **MY18B20** sensor.
I think this is a great idea! It would be very convenient to connect this sensor to the **TS0201 TZ3000** sensor, since it is powered by two AAA elements. 

Is it possible to get firmware for TS0201 TZ3000 with MY18B20 support?

pvvx, 2024-10-25:
The interface in TelinkMiFlasher.html is not yet ready to work with different types of sensors.

Thermometers can service a maximum of 2 MY18B20 sensors. But to reduce consumption, each MY18B20 is connected to different outputs - the poll is carried out in parallel.

Today there is only a firmware build for the TB-03F module
[TB03F_v48.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/TB03F_v48.bin)

[image]

pvvx, 2024-11-03:
Added beta [TS0201S.bin](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/TS0201S_v48.bin) - connecting MY18B20 to PD2 pin.
Temperature from MY18B20 is transmitted in BTHome format.

closed 2024-12-01 by pvvx

dkbiz83, 2025-04-03:
> Added beta version:
> 
> 1. One MY18B20 on PD2 [TS0201S1](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/TS0201S1_v48.bin), [ZTH01S1](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/ZTH01S1_v48.bin)
> 2. Two MY18B20 on PD2 and PD4 [TS0201S2](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/TS0201S2_v48.bin), [ZTH01S2](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/ZTH01S2_v48.bin)
> 
> Temperature from MY18B20 is transmitted in BTHome format.
> 
> [#571](https://github.com/pvvx/ATC_MiThermometer/issues/571)

Hi there,

I have been looking for TS0201, ZTH01 on Aliexpress to implement your [mod with external sensors](https://github.com/pvvx/ATC_MiThermometer/issues/571), however, I got ZG-227ZS v1.2 as pictured above. I was trying to find TS0201, ZTH01 again, but it looks like Chinese change the hardware too fast and there are no options left with the same TS0201/ZTH01 hardware. Is there a chance you could help add MY18B20 support to ZG-227ZS v1.2? I would really appreciate that!